eegsynth/eegsynth #4: implement virtual synthesizer

robertoostenveld, 2015-11-23:
Yesterday I came up with the idea of a virtual synthesizer for testing, using the audio output of the computer (either laptop or RPi) as output device. 

Today I have implemented a VCO with mixer, LFO, and VCA. They are in a fixed linear arrangement, but all control signals can be fed in from external through REDIS, e.g. from the LaunchControlXL or from the OpenBCI EEGSynth modules. 

It would be nice to extend the synthesizer with a ADSR and possibly a VCF, and to implement a sequencer that could be used to feed the synthesizer.

stephenwhitmarsh, 2015-11-26:
Wow had de laatate emails nog niet gelezen.
Net klaar met presentatie in Malmo - het going goed! Zowel praatje als
demo. We hadden uiteindelijk maar 30 min, maar waren goed in tijd.

robertoostenveld, 2015-11-26:
Goed te horen dat presentatie goed gong. 

 Gisteravond tot laat aan de ADSR gewerkt, was heel leuk! Ik heb nu best een complete virtuele synth waarmee ik redis en dergelijke heb getest. Ik wil nu nog een virtuele sequencer maken. 

Zet je definitieve slides in de dropbox?

Robert

closed 2016-02-24 by robertoostenveld