eegsynth/eegsynth #14: Documentation

stephenwhitmarsh, 2016-03-06:
Have to add install of redis

robertoostenveld, 2016-03-09:
@stephenwhitmarsh: I have made some edits to http://www.ouunpo.com/eegsynth/?page_id=163, including some fixes, replacing "EEG-synth" with "EEGsynth", and added the OutputOsc module.

robertoostenveld, 2016-03-10:
Regarding the style of the documentation, I think we should distinguish between audiences (conceptual versus technical). The technical documentation would in my opinion better fit closer to the code.

Regarding the location of the documentation, we can consider 
http://eegsynth.org
https://pages.github.com
or a wiki on the github eegsynth project (which is now disabled in the github settings). 

Furthermore, I think that http://eegsynth.org should be the proper domain and that it should not redirect to http://www.ouunpo.com/eegsynth (although it is hosted on the same server). That is probably a setting in wordpress. If we properly use the DNS domain eegsynth.org, we can also explicitly use www.eegsynth.org, blog.eegsynth.org, wiki.eegsynth.org, tech.eegsynth.org, etc. and link between the different sites.

robertoostenveld, 2016-03-14:
I went through the full installation once more, on a clean Raspberry Pi. Along the way, I have updated the main eegsynth/README.md and the eegsynth/interface/README.md documents.

robertoostenveld, 2016-04-09:
I have copied some of the documentation from the website to github. I also restructured, not starting with the installation details.

closed 2016-04-09 by robertoostenveld