eegsynth/eegsynth #35: Brain module: Allow frequency bands to be controlled by Redis

stephenwhitmarsh, 2016-03-20:
In the current implementation of the Brain module the frequency ranges cannot be controlled live, i.e.using the launchcontrol. This could be solved by defining upper and lower frequency of each band separately, right?

robertoostenveld, 2016-03-20:
yes, that would not be difficult to add. The reason for not doing it already is that I am worried about the accuracy of the sliders and the lack of feedback. It would be difficult to check what the band limits are and to fine adjust. 

I will implement this though, since the lack of feedback is not sufficient reason for not having this.

robertoostenveld, 2016-03-20:
the [scale] option in the outputosc.ini file already supports either values or strings and could serve as example w.r.t. code

The idea would be 

[band]
alpha=9-11
gamma=launchcontrol.control077-launchcontrol.control078

where in the 2nd case it detects it as strings and reads the corresponding values from redis.

The same type of transparent float/redis handling should be supported throughout and probably needs a helper function in lib/EEGsynth.py.

robertoostenveld, 2016-03-20:
I implemented a generic function for getting int/float values either from the ini, or (if specified there as a string) to get them from redis. The formatting of the item in the ini file should be like this
``` 
item=1-20         this returns [1,20]
item=1,2,3        this returns [1,2,3]
item=1,2,3,5-9    this returns [1,2,3,5,9], not [1,2,3,4,5,6,7,8,9]
item=key1,key2    get key1 and key2 from redis
item=key1,5       get key1 from redis
item=0,key2       get key2 from redis
```

robertoostenveld, 2016-03-20:
this solves it

mac011> git commit -a
[master f5b431f] improvements to getting int/float values either from the ini or from redis, use the new functoinality for frequency ranges in the brain module (see issue #35)
 4 files changed, 81 insertions(+), 32 deletions(-)
mac011> git push
X11 forwarding request failed on channel 0
Counting objects: 27, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (10/10), done.
Writing objects: 100% (10/10), 2.01 KiB | 0 bytes/s, done.
Total 10 (delta 6), reused 0 (delta 0)
remote: This repository moved. Please use the new location:
remote:   https://github.com/eegsynth/eegsynth.git
To git@github.com:robertoostenveld/eegsynth.git
   cdb4563..f5b431f  master -> master

Please see the updated brain.ini for an example

closed 2016-03-20 by robertoostenveld