eegsynth/eegsynth #36: add support for MIDI synchronization to volca's

robertoostenveld, 2016-03-20:
as suggested by @nicofarr 

Having this would mean that we don't need a TTL gate channel as an (analog) trigger.

The PDFs mention that the volca's support synchronization, but I do not understand yet how that would work with python->mido->portmidi

nicofarr, 2016-03-20:
MIDI clock (and more generally Transport signals, such as start, stop) are control signals with specific values, defined here: 

https://en.wikipedia.org/wiki/MIDI_beat_clock

and here a more complete list : 

https://www.midi.org/specifications/item/table-1-summary-of-midi-message 

So sending the control change 0xF8 , 24 times per quarternote, using mido, should do the trick ? (that's actually more general than synchronizing Volcas, it would synchronize any MIDI device).

nicofarr, 2016-03-20:
So a MIDI clock module in EEGsynth would just take tempo as an attribute, as well as the identity of the MIDI port to send it.

robertoostenveld, 2016-03-25:
Hi @nicofarr,

Thanks for the pointers! The new midi clock  implementation was actually not that hard, but it required  some architecture changes. I rewrote the "eegsynth/pulsegenerator" and renamed it to "eegsynth/synchronizer". The idea is still that it translates a continuous (heart rate or slider) control value into a regular beat.

It now supports three (possibly simultaneous) outputs:
1) analog gate or TTL trigger using the arduino hardware cvgate module
2) sending a note/trigger to redis
3) midi synchronization

In Paris we used option 1. Since it conflicted with the eegsynth.outputcvgate module that also outputs the EMG level, it required a hack for the performance there. Shortly thereafter I implemented 2 instead, as it is cleaner from a design perspective. 

Today I implemented 3, and realized that it is actually not so hard to do all three simultaneous. Using  the ini file you can now switch either 1, 2 or 3 on or off. Besides a fixed on/off assignment in the ini file, you can assign toggle buttons (control values that are either 0 or >0, i.e. 127) that enable/disable them dynamically.

Besides the heart rate (or the value from a launchcontrol slider) control, there is a continuous multiplier (from 1/4 to 4x) and offset control.  

The video at https://youtu.be/XA8l68W0gZU demonstrates the new implementation. Here I am looping though the internal volca sequencer synchronization and the midi synchronization, and switching the redis trigger on and off  (which gets converted by the eegsynth/volcabeats module to the bass note).

So the bass note and the midi clock synchronization are in parallel. Also the hardware serial cv/gate converter (the one you have) would be in sync.

closed 2016-03-25 by robertoostenveld