eegsynth/eegsynth #37: module: launchcontrol breaks at button presse

stephenwhitmarsh, 2016-03-21:
the following occurs on button press (bottom rows):

pi@raspberrypi:~/eegsynth/module/launchcontrol $ python launchcontrol.py
------ INPUT ------
Launch Control XL MIDI 1
Midi Through Port-0
------ OUTPUT ------
Launch Control XL MIDI 1
Midi Through Port-0
-------------------------
Connected to MIDI input
Connected to MIDI output
note_off channel=8 note=42 velocity=0 time=0
Traceback (most recent call last):
  File "launchcontrol.py", line 147, in <module>
    ledcolor(msg.note, state1color[status])
  File "launchcontrol.py", line 84, in ledcolor
    outputport.send(mido.Message('note_on', note=note, velocity=color, channel=midichannel))
  File "/usr/local/lib/python2.7/dist-packages/mido/messages.py", line 367, in __init__
    setattr(self, name, value)
  File "/usr/local/lib/python2.7/dist-packages/mido/messages.py", line 379, in __setattr__
    check_databyte(value)
  File "/usr/local/lib/python2.7/dist-packages/mido/messages.py", line 233, in check_databyte
    raise TypeError('data byte must be an integer')
TypeError: data byte must be an integer

closed 2016-03-21 by stephenwhitmarsh

robertoostenveld, 2016-03-21:
lachen zeg! 

Ik ben er net achter dat de TouchOSC app helemaal niet via een extra lading in het OCS pakketje met de TouchOSC MIDI Bridge communiceert. De App doet het via (native OSX en iOS) rtpMIDI met de computer. Dus ook zonder de Bridge app, maar alleen met de juiste handmatige instellingen van de OSX MIDI interface settings kan ik de MIDI messages van de touchOSC app zien binnenkomen. 

Nu nog uitzoeken hoe ik zelf bij die MIDI kan komen �

15 minuten later (nou ja, eigenlijk 30 minuten): het werkt. Ik kan zonder enige TouchOSC toestand vanuit redis AVmixer aansturen. 

LaunchControl hardware �MIDI�> EEGsynth/launccontrol �> redis -> EEGsynth/avmixer �> AVmixer

Ik kan trouwens (nu ik weet dat die bridge niet nodig is) ook alleen met de TouchOSC app MIDI uitlezen en naar redis sturen. Dus

TouchOSC App �MIDI�> EEGsynth/midiinput �> redis --> EEGsynth/avmixer �> AVmixer

En wat dus al kon is 

TouchOSC App �OSC*�> TouchOSC MIDI bridge �MIDI�> EEGsynth/midiinput �> redis --> EEGsynth/avmixer �> AVmixer

waarbij OSC* dus eigenlijk geen OSC is maar OSC+rtpMIDI.

Dus samengevat werkt

 EEGsynth/whatever �> redis -> EEGsynth/avmixer �> AVmixer

dus voor al onze modules.

nu naar bed!
Robert

stephenwhitmarsh, 2016-03-22:
Woohooo!!! 
Inderdaad lachen!