eegsynth/eegsynth #41: bug in buffer.sh?

stephenwhitmarsh, 2016-03-26:
pi@raspberrypi:~/eegsynth/module/buffer $ ./buffer.sh start
./buffer.sh: 23: ./buffer.sh: OPTIONS+= : not found
./buffer.sh: 24: ./buffer.sh: OPTIONS+=1972: not found

robertoostenveld, 2016-03-26:
probeer eerste regel eens te veranderen van

#!/bin/sh
in 
#!/bin/bash
of
#!/usr/bin/bash

Het juiste commando vind je met 
> which bash

Als dat werkt (aanpassing in script), probeer dan eens

#!/usr/bin/env bash

als het daarmee werkt, gaat het zowel op RPi en OSX

stephenwhitmarsh, 2016-03-26:
top, dat werkt!

closed 2016-03-26 by stephenwhitmarsh

robertoostenveld, 2016-03-26:
Heb je commit of pull request gedaan?

stephenwhitmarsh, 2016-03-26:
ah nee, vergeten. morgen, moet nu uit voor een drankje. elektrodes onverrichte zaken van mijn lijf plukken. matlab op mijn pc aan het installeren want bufferviewer werkt niet lekker, en 2 gig aan scipy en MNE aan het installeren voor de heartrate

stephenwhitmarsh, 2016-03-26:
wacht ik probeer het even

stephenwhitmarsh, 2016-03-26:
ah moet nog een branch maken eerst. pfff.. morgen weer