eegsynth/eegsynth #54: offset and data are not of consistent type

robertoostenveld, 2016-10-09:
I am getting this error with the record module on RPi

./../../lib/EDF.py:157: UserWarning: Value exceeds physical_max: 65469
  warnings.warn('Value exceeds physical_max: '+ str(max(raw)));
Traceback (most recent call last):
  File "./record.py", line 149, in <module>
    f.writeBlock(np.transpose(D))
  File "./../../lib/EDF.py", line 159, in writeBlock
    raw -= self.offset[i]  # FIXME I am not sure about the order of calibrate and offset
TypeError: Cannot cast ufunc subtract output from dtype('float64') to dtype('uint16') with casting rule 'same_kind'

The first (warning) is not relevant, the second (error) is. It seems the offset and raw data are of different types. It is not clear to me why this error only happens in python on rpi and not on osx.

stephenwhitmarsh, 2016-12-21:
Related to #68.

robertoostenveld, 2017-10-29:
I have recently been doing recordings without getting this error, so it seems to have been resolved.

closed 2017-10-29 by robertoostenveld