eegsynth/eegsynth #55: bin/install.sh should be documented

robertoostenveld, 2016-10-23:


deep-introspection, 2016-10-23:
Got caught this morning but want to work on that this afternoon !

robertoostenveld, 2016-10-23:
The idea is that the motivation should get more clear. Why all these dependencies? When are they needed (and when not)?

robertoostenveld, 2016-10-23:
the title would better be to overall improve https://github.com/eegsynth/eegsynth/blob/master/doc/installation.md

deep-introspection, 2016-10-23:
To me, the instructions for OSX are very confusing. Maybe we should come up with a proper structure and put an index at the beginning. Can we do internal links in the Markdown readme?

robertoostenveld, 2016-10-24:
yes, we can do internal links, they work pretty much the same as external ones. 

One issue with OSX is that I have never done a clean install. I have been developing on a single MacBook, abandoning certain attempts (and dependencies) and adopting others. Furthermore, I have (historically) used MacPorts, whereas nowadays I think that Homebrew is to be preferred as the "missing package manager". 

Not even fully clear right now is what the dependencies are, and where the dependencies are needed. E.g. for the online web interface you need Node JS (which is a bit of a messy installation on RPi, but clean on OSX), but you don't need it for anything else. With Python it is probably the other way around, i.e. default on RPi, but not trivial to get it right on OSX. 

Thinking about maintaining installation instructions, it occurs to me that it might be easier to develop them using virtual machines with various guest OSes. QEMU can emulate the RPi. With a collection of VMs I would be able to clone them and keep a clean version at all times.

deep-introspection, 2016-10-24:
Ok, I will try to do on my own a clean installation and document it!

Maybe the README structure could go as:
- **Introduction**, like what you did here: http://www.ouunpo.com/eegsynth/?page_id=514
- **Installation**
    - _Linux_ (Has it been tester outside the case of RasberryPI?)
    - _Mac OSX_
    - _Windows_
- **Roadmap / HowToParticipateInDev**

I strongly agree with the VM idea. We can also use containers like with Docker/Vagrant to be even more lightweight.

robertoostenveld, 2016-10-24:
good that you mention the website. @stephenwhitmarsh and I have been struggling with where to document. Some of it (the more static parts) are now on wordpress and some (the more dynamic parts) are here on github. We should avoid duplication, so perhaps only copy a small section over to the README file and point to the website for more details.

Having a roadmap and howtoparticipate is an excellent idea. I would suggest making those separate files, and have the files (in a common section at the top, or at the bottom) refer to each other.

robertoostenveld, 2016-10-24:
testing outside of RPi has not been done systematically, although I think that some of the participants this weekend had ubuntu laptops. 

This is actually what triggered me to think about VMs, as we might want to consider Debian/apt and Redhat/yum based distributions separately. 

A good initial test would already be to try installing it on a clean ubuntu or mint distro.

stephenwhitmarsh, 2016-10-25:
I'll install EEGsynth on my new Mint installation in a week or two, after conference/holiday

robertoostenveld, 2016-10-26:
@Elisabeth-github commented on slack that windows documentation is missing. Considering windows might actually help us to think of a general strategy to deal with different platforms. That does not mean that I'll document it in detail for windows, but thinking about how that documentation would look like is helpful.

deep-introspection, 2016-10-31:
I add some documentation for OS-X.
See the pull request: https://github.com/eegsynth/eegsynth/pull/59

stephenwhitmarsh, 2016-11-14:
Installed EEGsynth on my new Mint installation. It was painless and everything ran. 

So far one openbci2ft gave an error trying to access /dev/ttyUSB0, returning "Bad file descriptor". This is because it was readonly. Running openbci2ft with sudo solved this.

Today at CRI the right way of attaching electrodes, using enough gel, using alcohol, etc., was addressed. Would be good to have a couple of lines/pictures on this, probably linking to a website (openbci?). Same with referencing schemes.

robertoostenveld, 2016-11-15:
I added the USB error handling to the FAQ.

mac011> git commit -am "add Bad file descriptor error to FAQ" 
[master 2cbbca7] add Bad file descriptor error to FAQ
 1 file changed, 6 insertions(+)

mac011> git push
X11 forwarding request failed on channel 0
Counting objects: 4, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (4/4), done.
Writing objects: 100% (4/4), 704 bytes | 0 bytes/s, done.
Total 4 (delta 2), reused 0 (delta 0)
remote: Resolving deltas: 100% (2/2), completed with 2 local objects.
To github.com:eegsynth/eegsynth.git
   e654b25..2cbbca7  master -> master

robertoostenveld, 2016-11-15:
I would prefer if we can point to external documentation about electrode application. But if you want to write something yourself (with pictures!) that is also fine with me. I don't know whether you followed slack yesterday, but I posted this https://docs.google.com/document/d/1KiLyA4i5xjyxNqFW0hKOy6bOR4Ek0HaFaIVgrNWt-KQ/edit?usp=sharing which has some (suboptimal) photos that you could start with.

stephenwhitmarsh, 2016-12-21:
At this moment it seems the installation instructions are pretty up to date. One rather specific dependency arose installing it on OSX, i.e. #66, but I didnt find it too confusing. I worked on a very old, non-dev laptop, and was able to install dependencies, even starting with installing get-apt. etc. So I am closing this for now.

However, I will open a new issue on making a getting-started/tutorial on the level of: how do I plug in the OpenBCI dongle, and how do I attach electrodes. I am hoping that at the Brain Control Club they can help with that as several documents are already circulating.

closed 2016-12-21 by stephenwhitmarsh