eegsynth/eegsynth #60: Fieldtrip buffer time stamps

deep-introspection, 2016-10-31:
@robertoostenveld Are the time stamps of the Fieldtrip buffer taking into account the eventual loss of a packets (like with a clock)? # #

robertoostenveld, 2016-11-01:
hmm, let me think... The timestamps are acquisition device specific, i.e. not a feature of the buffer server but of the acquisition client. 

The only acquisition client that supports it (so far) is openbci2ft, but if I recall correctly the plan was/is to add it to more clients. I should look up the corresponding issue on bugzilla/github.

The code is here
https://github.com/fieldtrip/fieldtrip/blob/master/realtime/src/acquisition/openbci/openbci2ft.c#L673
which uses the timestamp.[ch] files in the same directory.

robertoostenveld, 2016-11-01:
It is not documented on bugzilla/github, only in emails with @stephenwhitmarsh. Those date from August 2015, as does the implementation for opebci2ft.

The emails also mention 
http://ebrain.io/openbci-ftdi-driver/
and
http://www.ftdichip.com/Support/Documents/AppNotes/AN232B-04_DataLatencyFlow.pdf
as OS X and WIndows solutions to improve timing for the FTDI driver that is used with the RF dongle.

robertoostenveld, 2016-11-01:
the openbci2ft and the jaga2ft acquisition clients are the latest and most modern. I wish I could find time to copy part of their new features (a.o. the ini file handling) to all others.

robertoostenveld, 2016-11-01:
@deep-introspection please clarify your initial question. 

The timestamps by themselves don't fix anything, but allow detection (and possibly correction) of jitter and delay. 

I did not plan or consider these timestamps for packet drops. Dropped packets in the RF communication cannot be detected, and failed FT buffer client-server calls will give a visible error.

robertoostenveld, 2016-11-01:
I have just also implemented timestamps for the jaga2ft acquisition client. To make this work, it now supports config/ini file parsing. Furthermore, I implemented the option for switching channels on/off and changing channel labels.

deep-introspection, 2016-11-01:
@robertoostenveld My initial question was about dropped packets in the RF communication so you answered. We could have for instance NaN values when the data stream is not coming anymore, and then keep the timing. Or, for each sample, have the clock/timer value assigned.

robertoostenveld, 2016-11-02:
distinguishing between dropped packets and delayed packets is not trivial based on timestamps, especially when you have to make that decision quickly. E.g., if you move your USB mouse while a RF packet arrives at the dongle, that packet will experience some delay along the USB stack to the software.   

if you look at https://github.com/fieldtrip/fieldtrip/blob/master/realtime/src/acquisition/openbci/openbci2ft.c#L702 there is a byte with the sample number. I presume that it will roll around every 255 samples, but that still allows packet-drop detection. Perhaps we should start by exposing that sample number as additional channel, i.e. add that as an option in the config file. It would be interesting to see how the unwrapped sample number compares against the timestamp.

robertoostenveld, 2016-11-05:
@deep-introspection I have added a "Sample" channel to the openbci data stream. It contains the sample as specified by the board itself. Since it is a uint8, it is between 0 and 255. I added an unwrap option to the source code (default=on) to have it continuously increment. I only used it for a short amount of time and have not yet seen any packet drops. 

The code of the binaries in is fieldtrip/realtime/bin/xxx where xxx is the platform.

Please update to the latest eegsynth version from github and run eegsynth/bin/install.sh and it should download the latest recompiled binaries from the FT github repository. 

I suggest that we close this issue for now, unless we see a need to implement some form of data insertion at the missing packets. But to work on that we should first identify that there are missing packets...

closed 2016-11-05 by robertoostenveld

deep-introspection, 2016-11-05:
Looks great! Thanks @robertoostenveld