eegsynth/eegsynth #62: you should be able to specify channels for recordctrl

robertoostenveld, 2016-11-19:
Right now it records whatever is in redis, including its own control buttons. Better would be that you could specify a filter on the channels to record.

stephenwhitmarsh, 2018-01-07:
Agreed, I ran into a related problem, where my Redis DB still contained old keys that were irrelevant and did not change. I had to flush it. Having keys to be recorded set in the ini would have prevented that problem.

stephenwhitmarsh, 2018-01-10:
@robertoostenveld Any thoughst on using wildcard in ini? Do you kano if they be easily used in Redis? Given our relative consistent postfixing, this would be quite convenient.

stephenwhitmarsh, 2018-01-17:
I would think that the same might be then consistently implemented in also the playback modules. For instance currently one cannot exclude channels for playback, which would help in testing and debugging.

robertoostenveld, 2018-01-17:
I think that the wildcards are a good idea.

We could use https://redis.io/commands/keys to get all keys, and then https://docs.python.org/2.7/library/fnmatch.html?highlight=fnmatch#module-fnmatch for the matching. Or use the wildcard in the keys() command directly. 

I wonder which modules would benefit from this. I guess all modules where there is no explicit mapping between python channels and redis channels, and where there is no explicit additional options per channel. E.g. postprocessing would not work. But plotcontrol would, right?