eegsynth/eegsynth #67: Make a check for connected MIDI devices

stephenwhitmarsh, 2016-11-28:
For instance in a /utils directory. For now one can use the Volca modules which output connected MIDI devices

robertoostenveld, 2016-11-28:
you mean to have a small utility that lists connected devices?

stephenwhitmarsh, 2016-11-28:
indeed

stephenwhitmarsh, 2016-12-04:
Interestingly, on my MINT installation, there is a program called amidi, which lists devices and can monitor. Perfect for the purpose of identifying MIDI device and mapping buttons. See: https://freesoftwaremusic.wordpress.com/2015/03/29/midi-monitors/comment-page-1/

stephenwhitmarsh, 2017-09-13:
Will be a very platform/use specific issue, and solvable with some googling.

robertoostenveld, 2017-09-13:
I suggest to add this to the doc/faq.md file, e.g. as "How can I identify my midi devices?"

All small pieces of documentation are valuable; documentation does not have to be complete from the start, but we should allow it to grow in small steps.

stephenwhitmarsh, 2018-12-11:
Great page with command line tools to explore soundcard and midi devices:
http://tedfelix.com/linux/linux-midi.html