eegsynth/eegsynth #68: Consider/describe/organize/standardize units/scales.

stephenwhitmarsh, 2016-12-17:
We should create a separate page where we describe the units and scales of different processes/modules and see if we can converge or at least make it as transparant as possible. Think MIDI 6/12 bits, OSC, CV voltages (1v/oct, 1v/db etc), launchcontrol units (0-127 cq midi), etc.

stephenwhitmarsh, 2016-12-18:
About the externals:

The EEGsynth uses the **Endorphins Cargo** optimally with the +/-5v Pitchweel setting. This represents the maximum range. For more info see: http://www.ouunpo.com/eegsynth/?p=480
Through the endorphins .ini this can be restricted to e.g. 0-5v. Because of the many (16) output channels, it is handy to test with both the polar and nonpolar outputs, the latter mainly for the LFO/PW/ECHO modules of the Erebus. 

The **Erebus Dreadbox** analogue synthesizer has the following specific patching specifications (from the manual: http://www.dreadbox-fx.com/wp-content/uploads/2016/04/erebus_manual_v2.pdf):

INPUT OSC pitch: +/- 12v, 1v/oct 
INPUT VCA: +/- 5v

INPUT LFO(rate): 0-5v
INPUT PW: 0-5v
INPUT Echo: 0-5v, best at 0-2.5v

OUTPUT OSC: 1v/oct, converted from MIDI if needed
OUTPUT ADSR: 0-6.6v, depending on depth setting
OUTPUT LFO: +/- 5v

**Microbrute** (probably) has the following specifications, according to a nice article here: 
http://www.hars.de/2016/01/microbrute-eurorack.html

INPUT/OUTPUT OSC pitch: 0-10v, 1v/oct
OUTPUT LFO: +/- 5v
OUTPUT ADSR: 0-4.5v

I found no further specifications on the other input patches, i.e. sub/PWM etc., but given that they are well-controlled by its own LFO, I would think they work well for +/-5v as well.

stephenwhitmarsh, 2016-12-20:
About file I/O:

The EDF format is used for playback of data (playback module) as well as control signals (playbackctrl module). Data in the EDF format has the int16 format, going from 0 to 32767. 

In MATLAB I write data that I first normalized between 0 and 1, as follows:
dat = int16(data_CV.trial{1}*32767);
ft_write_data(fname,dat,'header',hdr,'dataformat','edf');

In the EEGsynth modules, one can then transform these values into a new range, depending on the module/device. I will post next on the Endorphines output, and the serial2CVgate output.

robertoostenveld, 2016-12-20:
I think we should consider instead how to get the right calibration values on the fieldtrip writing side of things. There is no reason that the edf itself would not contain whatever value you like it to be.

commit 955f86f2 (2017-12-28): FIX #68 - added device specific documentation from issue to documentation
closed 2017-12-28 by robertoostenveld