eegsynth/eegsynth #69: playbackctrl does not play back in according to samplerate

stephenwhitmarsh, 2016-12-20:
The module needs a stress-test as well as a code-check I think. It seems that when playing back 15 channels at 20Hz, the module is not able to keep up at all - about 40% behind in 100second, if not more (tested on my Mint laptop). Better would be to drop packets rather than (increasingly) lagging behind.

In any case, the dependency with the Redis should be considered as well - what would be the fastest rate it can be read/written to? Will the number of channels, written/read  significantly reduce the ability to keep up?

A propos, 20Hz seems to be able to capture most EEG dynamics, but still gives a stepping arpeggio when used for pitch.

stephenwhitmarsh, 2016-12-20:
printing desired/elapsed given me values between 0.5 and 0.77. This is independent from the learning factor, which does not seem to have an effect. I guess we are at the limits at 20Hz. Still it is strange it never goes above 1.0, and neither does it, in the end, seem to go fast enough, which <1.0 does suggest...

robertoostenveld, 2016-12-20:
I should admit that I did not envision 20x15=300 updates (database transactions) per second when implementing.

The learning rule might be incorrect. I remember struggling with it.

Since all channels are sampled synchronously, it would make sense to use MSET https://redis.io/commands/mset instead of SET. That reduces the number of DB connections with 15x.

Dropping frames would also make sense. 

I think that these are three desired improvements that should all be implemented.

stephenwhitmarsh, 2016-12-20:
I see that the loop regularly takes up to 100 even 130ms. The learning rule is cool, but is made impossible to converge when desired < possible. Anyway, a division by 15 would be great indeed. I'll fool around more with it later.

I'll also take a closer look at the Endorphines and see how the delay in sampling redis is implemented there.

stephenwhitmarsh, 2016-12-21:
I implemented r.mset, which needed me to understand python dictionaries 👍 

Also improved (fixed) the learning of the calibration. Would be nice to adapt the learningrate (setting it to 0.05% seemed to work well), It is rather surprising how variable it is.

Dropping frames seems a bit of a hassle at the moment and not necessary, or maybe even desirable, given that it is very hard to estimate the latency, and also that it seems to work fast enough now - tested with 15channels on 20Hz.

closed 2016-12-21 by stephenwhitmarsh

robertoostenveld, 2016-12-23:
if you implement changes related to the code, please mention them in the commit comment. That is done by prefixing it with the hash symbol, e.g. #69

If you do that, the relevant commit will show up in the discussion of the issue.

stephenwhitmarsh, 2016-12-23:
noted :-)