eegsynth/eegsynth #71: Reflection on modules and their description

stephenwhitmarsh, 2016-12-22:
At the moment we lack an explicit way - such as in the module name or on a separate page, or even in the ini file - to reflect where in the EEGsynth architecture the module need to be patched. In fact, we might need to consider the patch-ability of a module from the point of view of itself as a module, in relationship to other modules and devices. One way to go it to define them with regard to each other in some axioms:

**Axioms**
A1) Modules are never directly patched with each other, always via Redis, e.g.:
_Module <--> Redis <--> Module_

A2) Devices never connect directly to Redis, always via Modules, e.g.:
_Device --> Module --> Redis_

A3) Multiple modules can interact with Redis asynchronously. 
_Module --> Redis_
_Module ------^_

A4) Multiple modules can read from a single device directly, except serial/MIDI devices:
_Device --> Module
       |-----> Module

A5) Multiple modules cannot write to a single device (except perhaps for the extra postprocessing buffer, but not relevant yet)

**Classifications**
Within these axioms, there are different classifications of the modules.

C1) Some modules only read and return to Redis, such as the postprocessing module:
_Redis <--> Module_

C2) Some modules sample a device and return to Redis, such as the Brain or Muscle module:
_Device --> Module --> Redis_  

C2') In fact, some modules can also read from the Redis, e.g. for analysis parameters:
_Device --> Module <--> Redis_  

C3) Some modules sample a device and return to Redis, and return to the device as well, such as the LaunchcontrolXL or Volcabeats:
_Device <--> Module --> Redis_
 
C3') In fact, some modules can also read from Redis, e.g. for analysis parameters:
_Device <--> Module <--> Redis_ 

Now, if we consider ft_openbci2ft and the ft_buffer _together_ as a _virtual device_, this would allow it to be part of C2, i.e. as a nonwritable device:
C4) _[Device --> Module]_ 

We can then consider the preprocessing module and the (extra) buffer together as a combination of a module and a writable virtual device:
C5) _[Device <--> Module]_

Okay, some things need to be clarified, and I wonder how much clearer it actually makes it. Perhaps it is better to create in the readme of each module their function, purpose and part in a patch, standardized according to the above.

robertoostenveld, 2016-12-23:
I think of the modules like this

[image]

This maps the primary input and output. Some modules also have a secondary input to switch the function on or off, e.g. to toggle recording or playback.

I think you are right with the Axioms, except for A5. I believe it is possible to have two modules output to a single midi device. Also the usb2cvgate with serial-over-usb I recall that to work (unexpectedly), i.e. one module sending gates and the other module sending the control values.

If you look at my mapping excel file, you see the "devices" that I recognize. It also lists "file", which is actually the "disk" device. But for most I don't mention the device itself, but the interface (e.g. midi or usb). 

Since I have 6 columns, where there is 1 out of 3 choices for the input, and 1 out of 3 choices for the output, I would expect that there are 9 possible module classifications. Here I am also explicit about the buffer, since there are also the record and playback module, and two explicit hardware devices now (but more in the future). 

Actually, a signalgenerator module (simulated EEG) would make sense, using the launchcontrol sliders to change the signal.  That is redis--->module--->buffer.

Looking back at your suggestion: The axioms make sense, your classification makes less sense to me.

robertoostenveld, 2016-12-23:
here another copy, after some fixes and after sorting on all 6 columns

[image]

robertoostenveld, 2016-12-23:
interesting if you look at it like this is that you can classify them in blocks. Not all blocks have a clear label, but there are signal processing modules, output modules, modifier modules, input modules, etc.

stephenwhitmarsh, 2016-12-23:
I like it!!!

<https://cloud.githubusercontent.com/assets/899043/21451833/810540ee-c902-11e6-971c-08a235bccec3.png>

or mute
the thread

stephenwhitmarsh, 2017-09-12:
Robert, if you still have this excel sheet, I will put it on the site.

robertoostenveld, 2017-09-13:
I don't think we should upload excel files. Better put it online as a proper table, that allows us to edit. This works for me on my homepage:
https://wpbtips.wordpress.com/2013/04/09/table-coding-for-wordpress-com-users/

I don't have the table on this computer, but somewhere at home. Probably easier to start from scratch, also to ensure it is up to date. 

where do you want that page?