eegsynth/eegsynth #72: Rename Brain and Muscle modules

stephenwhitmarsh, 2016-12-22:
To remain consistent with considerations of modularity of the modules (see #71), we should rename the Brain and Muscle modules according to their function, not their part in the patch. I suggest we rename:
Brain --> FFT
Muscle --> RMS

robertoostenveld, 2016-12-23:
I suggest to call then spectralpower and envelope. The name should reflect their function, not the internal algorithm.

stephenwhitmarsh, 2016-12-23:
yes, agreed, but "envelope" is unclear.

wrote:
>
> To remain consistent with considerations of modularity of the modules
(see #71 <https://github.com/eegsynth/eegsynth/issues/71>), we should
rename the Brain and Muscle modules according to their function, not their
part in the patch. I suggest we rename:
> Brain --> FFT
> Muscle --> RMS
https://github.com/eegsynth/eegsynth/issues/72>, or mute the thread <
https://github.com/
52KD0XL_Jks5rKps4gaJpZM4LUGie>.

or mute
the thread

robertoostenveld, 2016-12-23:
✉️

robertoostenveld, 2017-10-29:
since there is no progress on this issue, let me close it.

closed 2017-10-29 by robertoostenveld