eegsynth/eegsynth #80: Trouble with the Mac64 openbci2ft binary

deep-introspection, 2017-08-03:
Apparently, the new version of the openbci2ft binary from Fieldtrip is not working on Mac64
It gives always a "Segmentation fault: 11" when run directly on the binary, and gives no feedback at all when started from the shell module.

I download the previous version (before Daisy board support), with the commit https://github.com/fieldtrip/fieldtrip/tree/49f7d56cd468ed220fae1d4d54affc56a9531ab5/realtime/bin/maci64

Seems to work with this version.

Note: @stephenwhitmarsh 's Linux Mint version is working well.

robertoostenveld, 2017-09-13:
what is the ini file configuration that you used?

when I do

```
cd eegsynth/bin
./instal.sh
./openbci2ft
```
I get
```
Use as
   openbci2ft <device> [ftHost] [ftPort]
with the parameters as specified below, or
   openbci2ft <config>
with the name of a configuration file for detailed setup.

When ftPort is omitted, it will default to 1972.
When ftHost is omitted, it will default to '-'.
Using '-' for the buffer hostname (ftHost) starts a local buffer on the given port (ftPort).

Example use:
   openbci2ft /dev/tty.usbserial-DN0094FY                 # start a local buffer on port 1972
   openbci2ft /dev/tty.usbserial-DN0094FY - 1234          # start a local buffer on port 1234
   openbci2ft /dev/tty.usbserial-DN0094FY serverpc 1234   # connect to remote buffer running on server PC
```

can you confirm this works? If not, it might be a shared library issue.

robertoostenveld, 2017-10-29:
I guess this issue has been solved. If not, please reopen.

closed 2017-10-29 by robertoostenveld

deep-introspection, 2017-10-29:
The problem is that every helmet needs a specific address.
How can we make it adaptable without interference with git?

stephenwhitmarsh, 2017-10-30:
Robert and I discussed the fact that every use will need different ini
files. Robert updated all modules that now need to be executed either with
the default ini (in the module directory) or by using -i [inifile].
Best practice is to create a directory which will not be tracked by git,
but in which you store your own ini files and call those.
Also think about multiple directories for multiple 'patches'.
Related, the shell scripts have been removed (except when the module is an
executable).
In the near future, i.e. for COGITO, this would allow us to create a cogito
directory with the cogito ini files (different for each Pi/laptop), and a
shell script which starts them up in the right order and with their own ini
files.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

robertoostenveld, 2017-10-31:
and we can share a particular patch, by sending a zip file with some ini's and a README that explains which modules need to be started up (using those ini's).