eegsynth/eegsynth #88: Windows installation documentation

stephenwhitmarsh, 2017-09-13:
Installation instructions have been organized more clearly in win/linux/osx, making clear we do not have any instructions for windows. @robertoostenveld you just installed the EEGsynth on a windows 10 system, did you make any notes of how you did that?

robertoostenveld, 2017-09-13:
There are at least 4 ways this can be done (with sub-options for each of them). I tried most of them, although not all of them exhaustively, running into various (different) problems for each.

1) native python, using anaconda
2) cygwin, using pip
2a) 32 bit version
2a) 64 bit version
3) WSL, using apt and pip
4) in virtual box on a linux VM, using apt and pip

1 can probably also be done with pip, 2 and 3 and 4 can certainly also be done with anaconda. 

Since non of them worked 100% so far, I don't know which is the preferred method that we should document. 

PS I hope you can imagine me chasing dependencies, apt-get'ing, pip'ing and constantly running into problems for about 3 full evenings in a row...

stephenwhitmarsh, 2017-09-13:
I can definitely imagine... especially the pain not having either option work satisfactory...

closed 2017-10-03 by robertoostenveld