eegsynth/eegsynth #89: Support for OpenBCI Ganglion board?

ESchae, 2017-10-02:
Hello,
am I right that the openbci2ft module is meant to be used with the [OpenBCI Cyton Board](https://shop.openbci.com/collections/frontpage/products/cyton-biosensing-board-8-channel?variant=38958638542) which comes with an Bluetooth RF interface? Or is it possible to use it with the [OpenBCI Ganglion Board](https://shop.openbci.com/collections/frontpage/products/pre-order-ganglion-board?variant=13461804483) as well?
Thanks in advance for your help!
Elke

robertoostenveld, 2017-10-03:
it is indeed for the cyton board (which used to be called the "openbci" board).

Comparing http://docs.openbci.com/OpenBCI%20Software/04-OpenBCI_Cyton_SDK
with http://docs.openbci.com/OpenBCI%20Software/06-OpenBCI_Ganglion_SDK
it seems that many commands are similar.

At the time of writing the openbci2ft code (which is maintained in fieldtrip), there was only version 1 of the board and firmware. Since then, v2 and v3 of the firmware came out, the 16 channel version was added, and the wifi shield was and the Ganglion board were introduced.

If I could borrow a Ganglion board for a few weeks, I could test it and add support for it. I don't think that it needs a new implementation, some changes to the existing code would suffice.

ESchae, 2017-10-05:
Would these changes only have to be in [https://github.com/fieldtrip/fieldtrip/blob/master/realtime/src/acquisition/openbci/openbci2ft.c](https://github.com/fieldtrip/fieldtrip/blob/master/realtime/src/acquisition/openbci/openbci2ft.c) or also in other files?

robertoostenveld, 2017-10-06:
it would indeed only require changes to openbci2ft.c. 

In my opinion best would be to start with extending the configuration file. An example ini/txt file is at https://github.com/fieldtrip/fieldtrip/blob/master/realtime/bin/config/openbci_bipolar.txt
There is also a unipolar example. 

It now contains the option "daisy" with value "on/off". I think better would be to have the option "board" with allowed values "ganglion/cyton/daisy". Then depending on the board type, the code could act slightly differently (as it does now already for the daisy). 

Regarding the serial port: I usually look at /dev/tty* prior to plugging in, and then once more after plugging in. There should be a new device after plugging in which is the serial interface. But I don't know how the BT pairing would work on Linux.

robertoostenveld, 2018-11-26:
FYI I have just ordered a ganglion board. Once it arrives I will implement support for it.

robertoostenveld, 2019-09-01:
The OpenBCI ganglion board is now supported, although not directly through a python module. Instead, you can use the OpenBCI GUI, set up LSL in the GUI, and use the python lsl2ft module (in EEGsynth) to copy the data from LSL to the buffer. From there on it works as usual. 

This was for example used at the Tradgarden performance, see https://github.com/eegsynth/eegsynth/tree/master/patches/transformation/tradgarden for the full patch.

closed 2019-09-01 by robertoostenveld