eegsynth/eegsynth #100: document hardware more consistently

robertoostenveld, 2017-10-23:
We now have some (limited) documentation on the website. I propose that we create a hardware directory to the eegsynth repo with the photo's and more details. It will be easier to maintain and extend here than on the website.

See also #74

robertoostenveld, 2017-10-23:
@stephenwhitmarsh agreed?

stephenwhitmarsh, 2017-10-26:
Agreed, although we could consider putting the hardware in the appropriate module, e.g. the outputCVgate and DMX module. That would remove the need to create separate documentation as well and would increase understanding of the software-hardware dependencies.

robertoostenveld, 2017-11-04:
I have added documentation for the artnet2dmx hardware.

robertoostenveld, 2017-11-04:
I have now also provided more documentation for artnet2neopixel hardware. I think that completes it for now.

closed 2017-11-04 by robertoostenveld