eegsynth/eegsynth #99: move interface code out of the way

robertoostenveld, 2017-10-23:
As discussed with @stephenwhitmarsh on the train back from Dwingeloo: The bash scripts and web user-interface code is now rather specific and not that helpful when developping modules or when writing a new patch. 

I propose that we move the web interface and corresponding bash scripts (i.e. the ones that work like "module start|stop") to a separate repository. 

The python scripts can be started directly from the command line. The modules that are based on a binary (e.g. openbci2ft and ft2audio) are better implemented in a bash script that does not push it to the background.

Note that this also means that the script for redis as just updated by @deep-introspection in #94 will be moved to the other repository.

Could you please let me know whether you agree?

robertoostenveld, 2017-10-23:
my idea is to organize it such that 

```
eegsynth (this repo)
eegsynth-webinterface
eegsynth-webinterface/bin
eegsynth-webinterface/node
eegsynth-webinterface/doc
```

where bin contains the module start-up scripts and node contains the node.js web server code.

robertoostenveld, 2017-10-23:
for current users and developers of new patches the eegsynth-webinterface repository would not be of much relevance (if any)

robertoostenveld, 2017-10-28:
rather than making a new repository, I just moved all bash scripts to eegsynth/interface/scripts. We can move them elsewhere later. 

I also added the -i and the --inifile option to all python files.

robertoostenveld, 2017-10-28:
the following modules are not based on a python script and currently don't have a startup method.

```
buffer
jaga2ft
openbci2ft
outputaudio
redis
```

robertoostenveld, 2017-10-28:
I have removed jaga2ft altogether, as I am the only one with this hardware.

robertoostenveld, 2017-10-29:
I have send a message on slack to notify everyone of this change.

closed 2017-10-29 by robertoostenveld

deep-introspection, 2017-10-29:
Nice! So how shall we start redis then? Still with the shell script?

robertoostenveld, 2017-10-31:
there is a new shell script. It parses the ini file (which only contains the port for now) and starts redis in the foreground. Note that for linux (raspbian) the redis server is started at boot up as a daemon, so the script is only relevant now for OS X.

```
#!/bin/bash
#
# This will start the REDIS server according to the settings
# in the specified or the default inifile.
#
# Use as
#   redis.sh [-i <inifile>] [-h] [-v]

# include library with helper functions
. "$(dirname "$0")/../../lib/EEGsynth.sh"

DIR=`dirname "$0"`
NAME=`basename "$0" .sh`
BINDIR=$DIR/../../bin

# set the default
INIFILE=${DIR}/${NAME}.ini
VERBOSE=0

if [ -e "/usr/bin/redis-server" ]; then
  # on raspberry pi
  COMMAND="/usr/bin/redis-server"
else
  # on maci64
  COMMAND=`which redis-server`
fi

while getopts "hvi:" option; do
  case "${option}" in
    i)
      INIFILE=${OPTARG}
      ;;
    v)
      VERBOSE=1
      ;;
    h)
      echo "Use as: $0 [-i <inifile>] [-h] [-v]"
      ;;
    \?)
      echo "Invalid option: -${OPTARG}" >&2
      ;;
  esac
done

# this parses the ini file and creates local variables
shini_parse $INIFILE
PORT=$ini_redis_port

if [ ${VERBOSE} == 1 ] ; then
  echo INIFILE=$INIFILE
  echo PORT=$PORT
fi

${COMMAND} --port ${PORT}
```

My idea is to make all bash scripts to start the binaries as similar as possible. We now have 

module/buffer/buffer.sh
module/openbci2ft/openbci2ft.sh
module/outputaudio/outputaudio.sh
module/redis/redis.sh

The buffer one is a bit different than the others, since it uses eegsynth/bin/parallel to start multiple buffers (i.e. multiple executables) in parallel (e.g. on 1972, 1973 and 1974).