eegsynth/eegsynth #121: calibration should be improved

robertoostenveld, 2017-11-19:
It should be more user-friendly, more robust and more consistent over all modules. The patching that is required for calibration should be clearly documented. 

Manual calibration involves the launchtcontrol. Automatic calibration involves some historic (sliding window) channel statistics.

@stephenwhitmarsh  started working on this in the plotcontrol module, I started working on an alternative implementation in the new smoothing and calibration modules. Part of the functionality between plotcontrol and smoothing is now replicated. 

This needs to be evaluated and disucssed.

stephenwhitmarsh, 2017-11-19:
Skype tomorrow?

robertoostenveld, 2017-11-20:
I just pushed the new calibration module to github.

robertoostenveld, 2017-11-20:
for testing purposes it would help to have a generatecontrol module, just like the generatesignal. This would be very similar to an LFO, but could also include adding Gaussian and uniform noise.

robertoostenveld, 2017-11-20:
@stephenwhitmarsh I have just committed a simple generatecontrol module. It makes sine/square/triangle/sawtooth waves, with a (launchcontrol) configurable scale, frequency, offset, noise, etc.  

[image]

robertoostenveld, 2017-11-20:
the launchcontrol + generatecontrol + plotcontrol module together should provide the ideal testing ground for online calibration.

robertoostenveld, 2017-11-21:
we should consider how this fits with the post processing module, which now claims to also be doing calibration. Some replication over modules is not a problem, but we should not replicate too much.

robertoostenveld, 2017-12-28:
calibration is now documented

closed 2017-12-28 by robertoostenveld