eegsynth/eegsynth #122: make patching more flexible

robertoostenveld, 2017-11-20:
we now have 

```
[section]
option1=value
option2=launchcontrol.channel077
```

which implement direct fixed and flexible control values. Imagine that we would extend it like this

```
option3=somekey
```

where somekey would (in REDIS) contain the string value launchcontrol.channel077. 

This would allow a redirection of the control values to something that is specified in REDIS and makes the patching configurable in REDIS. We could even make a simple module that would only serve to inject some specified patching into REDIS.

stephenwhitmarsh, 2017-11-20:
exactly, this would make it generically possible for modules to reference
to the same higher-level description. Question is where this would be set.
Could be by several modules. This might create a conflict or unexpected
behavior. The first time it's defined or the last time? I.e. can it be
overwritten? I think it has to be because Redis does not flush after bootup
(but we could build that in). Also, although we discussed it earlier and
decided against it at that point - this does allows a higher-level above
module ini files.

In any case, how should we think about this being implemented without
hardcoding a list of recognized first-order keys, which is untenable. IT
makes sense instead to name the second.order keys something like
'redis.somekey'. If such is encountered, the EEGsynth.getfloat will be
called again on its redis value.

robertoostenveld, 2017-11-21:
I see the option of either giving all modules the possibility to export their configuration, e.g.

```
[general]
publish=yes ; can be yes/no
```

which then results in redis keys like

```
launchcontrol.midi.device=Launch Control
launchcontrol.midi.channel=9
```

and in general the pattern

```
module.section.item
```

I also see the (not persee conflicting) option that there would be one generic module that would write a whole bunch of options to redis.

The question you raise is whether options in redis should be overwritten. 

The questions I have is what is the precedence of options. Would the ones in redis overrule the ones in the ini-file or the other way around? And how would be specified that the option is to be fetched from redis? There is an issue with potential recursion.

robertoostenveld, 2017-12-27:
it would already help to make the code more consistent. The code in the modules now differs between 

1) getting it from the ini file
2) getting it from redis
3) getting it from either one (using lib/EEGsynth.py) 

I will make a "patch" class that incorporates all three of them. All modules should use this patch class rather than directly interacting with the ini file or with redis.

robertoostenveld, 2017-12-27:
the code is now consistent, but needs some serious testing

closed 2017-12-27 by robertoostenveld