eegsynth/eegsynth #123: 4-channel usb2cvgate hardware has errors

robertoostenveld, 2017-11-21:
... as already suspected after @stephenwhitmarsh fried multiple Nano's. I started soldering a new version, and after connecting the MCP4822 to the Nano there was a large drop of supply voltage (from ~5V to 3.4V) and the MCP got really hot. I figured out that 

- Vdd (5V) and Vss (0V) of the MCP4822 should be connected the other way around.
- Pin 4 of the MCP4822 should be connected to D11 (MOSI), not to D12 (MISO).

stephenwhitmarsh, 2017-11-27:
JESUS!!! 
Hahaha, so it might not have been me??? (At least not with regard to following instructions)...
So happy to hear this :-)

robertoostenveld, 2017-11-27:
I only assembled it partially and then proceeded with testing (as I realized your hardships). I got one very hot (on the touch) MCP4822 chip that did not output the correct voltage. I have tried it with the correct wiring on a breadboard, neither the nano nor mcp4822 are permanently damaged: there it worked as it should (with 4V output, since without the 2.5x amplification).

Could you send me the pre-drilled enclusure and components that you have? I will then finalize it.

stephenwhitmarsh, 2017-11-27:
Sure, I'll try to do that tomorrow.

stephenwhitmarsh, 2018-01-08:
Sorry, I forgot. I will do so now.

robertoostenveld, 2018-02-27:
From the top of my head I recall from the previous debugging session that the V+ and GND for the MCP4822 are swapped, and that something was wrong with the MISO and MOSI (connections to D12 and D13 on the Nano). The enclosure is now on its way to me, so I will soon assemble a new one and update the documentation.

robertoostenveld, 2018-03-03:
the recollection in my previous message was correct. I have swapped the Vdd (+) and Vss (-) in the design, and connected the SDI (slave-data-in) of the MCP4822 to the MOSI (master-out-slave-in) of the Nano.

The PDF has been updated in the repo with commit 03e52d0.

closed 2018-03-03 by robertoostenveld