eegsynth/eegsynth #124: disambiguate feedback from buffer module

stephenwhitmarsh, 2017-11-23:
The buffer module now returns feedback saying something like 'starting buffer 1972'. It would be good if also report whether or not it did so or not. It used to also return to the prompt, in the sense that it ran in the background. Is there a reason (i.e. because of the monitoring of errors throughout the chain that we discussed earlier) that this is the case?

robertoostenveld, 2017-11-23:
The startup script prints this just before it starts the buffer executable. The script itself does not check for possible errors. But possible errors will be printed on the screen (e.g. the port already being occupied because you started the buffer also in another terminal).

Try this in a terminal
```
mac011> ./buffer.sh 
Starting FieldTrip buffer on port 1973... 
Starting FieldTrip buffer on port 1972... 
```
and this in another
```
mac011> ./buffer.sh 
Starting FieldTrip buffer on port 1972... 
Starting FieldTrip buffer on port 1973... 
tcpserver bind: Address already in use
tcpserver bind: Address already in use
```
it shows the error and the script actually stops. 

If you go back to the first terminal, you can press ctrl-c to stop the first two buffers.

stephenwhitmarsh, 2017-11-23:
Bon. D'accord.

closed 2017-11-23 by stephenwhitmarsh