eegsynth/eegsynth #125: Disambiguate/update OutputOSC input/output ini

stephenwhitmarsh, 2017-11-23:
In outputosc, In the default ini the same keys are used in separate input and output fields. It would be more consistent to output directly from Redis. 

In fact, one should then remove the postprocessing options, also to stay consistent with our ideas now concerning, right? I.e. so far all processing of control signals are done by separate modules. Scaling and offset should then also be in included in postprocessing with settings in a separate ini field.

Agreed?

robertoostenveld, 2017-11-23:
agreed wrt compressor_expander, scale and offset. The same applies here 

```
mac011> grep compressor_expander */*.ini
calibration/calibration.ini:[compressor_expander]
endorphines/endorphines.ini:[compressor_expander]
outputartnet/outputartnet.ini:[compressor_expander]
outputcvgate/outputcvgate.ini:[compressor_expander]
outputdmx512/outputdmx512.ini:[compressor_expander]
outputosc/outputosc.ini:[compressor_expander]
```

For the input-output mapping I don't see how that would be done differently. I think we cannot specify a key in an ini file with periods or slashes in the name... although, a quick test suggests that that might actually work. A clean implementation would then have an ini file like

```
[general]
delay=0.05
debug=2

[redis]
hostname=localhost
port=6379

[osc]
; this is the address and port of the receiving software, i.e. this can be running remotely
hostname=localhost
port=8000

[output]
/1/faderA=launchcontrol.control077
/1/faderB=launchcontrol.control078
/1/faderC=launchcontrol.control079
```

and not have any input section (nor anything what the calibration module can do).

stephenwhitmarsh, 2017-11-23:
I think that would be it, indeed. Agreed?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>