eegsynth/eegsynth #126: the calibration and postprocessing ini are not consistent

robertoostenveld, 2017-11-23:
although the two modules are doing almost the same (albeit that postprocessing is more powerful). The postprocessing ini specification is more similar with outputosc, the calibration ini is more similar to plotcontrol.

The issue I see with the calibration.ini is that I would not know how to extend it to support the specification of different compressor_expander options for different channels. Perhaps we should consider including a separate section for each channel, e.g.

```
[input]
; control values to calibrate, separated by comma
channels=launchcontrol.control077,launchcontrol.control078

[output]
; the results will be written to redis as "calib.channel1" etc.
prefix=calib

[launchcontrol.control077]
lo=launchcontrol.control049
hi=launchcontrol.control050
scale=1
offset=0
```

we could also have a general section, with the default options that apply to all channels unless otherwise specified per channel.

stephenwhitmarsh, 2017-11-23:
I was writing down another option, when I realized it would actually be the
same :-)

But to be clear, it could also be something like this:

postprocessing.ini:

[compressor]
channels=launchcontrol.control077,launchcontrol.control078
midpoint=10 ; whatever parameters for the compressor
prefix=comp

[offset]
channels=launchcontrol.control077.comp
value=10

So in a way, each element of postprocessing could have it's own input, and
in some cases, output (prefixed).

compressor=launchcontrol.control077,launchcontrol.control078

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

robertoostenveld, 2017-11-23:
how would you in this case specify an offset of 15 for launchcontrol.control078 (while using 10 for launchcontrol.control077)?

stephenwhitmarsh, 2017-11-23:
Perhaps like this then?

[compressor]
channels=launchcontrol.control077,launchcontrol.control078
midpoints=10,20 ; whatever parameters for the compressor
prefix=comp

[offset]
channels=launchcontrol.control077.comp,launchcontrol.control078.comp
values=10,20

stephenwhitmarsh, 2017-11-23:
Not a fan of these comma-delimited long lists though

robertoostenveld, 2017-11-24:
Pfew, glad to hear that from you. You introduced that in the plot modules and I was afraid to complain about the comma-delimited lists 😉

stephenwhitmarsh, 2017-11-24:
Thanks for considering my feelings :-). But, yeah, can't solve all the world's problems at once ;-)
I do find something like:
channel1 = ...
channel2 = ...
..also ugly, specifically the use of the word 'channel' which is confusing in the context of (EEG) signal processing. It confuses raw data and control signals. Although a bit contrived, being consistent over modules/processes in using something like:
cv1 = ...
cv2 = ...
or perhaps better:
input1 = ...
input2 = ...
That still introduces a nagging feeling (why do I talk about feelings now?) that the actual number should matter, rather than the order of things, i.e. what would happen is they are inconsistent? Furthermore, the actual name, be it cv1 or input1, doesn't matter, because, as we discussed, they would not enter into the output/redis, i.e. it would be launchcontrol.note47.compressor, rather than cv1.compressor. At least, I still believe that the first option would make the patch most traceble throughout the pipeline.
In short, what other option do we have, i.e. how can lists be implemented in the config/ini structure?

stephenwhitmarsh, 2018-01-07:
Has been clarified with recent updates and decisions.

closed 2018-01-07 by stephenwhitmarsh