eegsynth/eegsynth #127: Define namespace (including 'units')

stephenwhitmarsh, 2017-11-27:
1) As discussed with @robertoostenveld, with increasing Redis traffic with the new modules (plot/calibration/postprocessing), it becomes important to decide on a more formal namespace. In fact, several related issues converge on this point:

2) #68: It might be able to disambiguate in what 'format' or 'units' the values are, e.g. MIDI, OSC, Volts (CVout), power, etc. In fact, I think we can make an informed overview of the different 'formats' we have or will support in the near future. 

3) Namespace might relate subsequent processes hierarchically, allowing them to be plotted in an organized way (e.g. with plotcontrol).  Once example: if one uses postprocessing to add an offset to a spectral estimate of alpha, currently things will work something like this (every module has their own prefix to key):
- spectral.alpha
- postprocessing.alpha_offset
Instead every module might append to existing key:
- spectral.alpha
- spectral.alpha.offset

Its not obvious to me at the moment of writing how 2 and 3 will interact.
4) #122: Furthermore, as discussed recently, patching might be made more flexible by allowing _values_ (of parameters) within a module to be send to Redis as _keys_. 

5) #126: We are as of yet undecided on how to select individual channels / all channels in e.g. postprocessing. In fact, when we have a consistent hierarchical namespace, I can image we could specify as channel either a low or high level label, i.e. if we want to specify a separate channel we could do:
launchcontrol.note045, or launchcontrol for all those starting with launchcontrol. Although a bad example, this might allow us to create groups, e.g. alpha.control.max, alpha.control.min could be referred to as alpha.control. Pfff... at second thought, let's abandon that for now :-)

stephenwhitmarsh, 2017-11-27:
Okay, this is my first suggestion, which is growing on me.

First, what are our different signals/formats used by the modules? This is definitely a loose definition of 'input'/'output' signals, but here goes:

- speed (accelerometer)
- mvolts (spectral/rms/generatesignal)
- bpm (heartrate)
- midi (launchcontrol/volca/keyboard/etc.)
- dmx (outputdmx/artnet)
- undefined control signals (generatecontrol, AVmixer) but see below
- undefined control events (sequencer) but see below

Those modules that expect midi (0-127) as manual input (most do) should just do so. If an alternative input needs to be used, the postprocessing module can deal with that. 

1) Each key starts with the 'format' of the value, e.g. 'mvolt','time','midi' and 'osc'. More about midi later (4).
2) Each module suffixes it's output, e.g. after calibration: spectral.min, spectral.max, spectral.cal
3) Postprocessing.ini has a field before and after other operations in which it can recode/rename redis keys, in addition to applying expressions. This will also be used for conversion between 'formats'. E.g. if one wants to create a MIDI signal from a calibrated spectral estimate, one can add at the end of postprocessing: midi.alpha = spectral.alpha.cal * 127 + 0.5 [rounding/casting to int might be done in output modules, as it knows it is midi by the key, or, if implemented, in postprocessing]
4) I see no real use of the different MIDI modules, i.e. the Volca, or even Launchcontrol. Correct me if I'm wrong, but a generic inputmidi and outputmidi module could suffice. I mean, that is the point of MIDI. The quantizing module can also just work on midi.
5) I find the dots (.) versus slashes (/) more consistent with namespace conventions in programming (https://en.wikipedia.org/wiki/Namespace). The outputosc module can replace dots with slashes. In the case we make an inputosc we can replace slashed with dots.
6) For now I would not yet cast/send values as keys. It is difficult enough. To solve the issue of plotting the relevant parameters in postcontrol with the relevant pre/post-calibrated signal, the calibration module can add the calibration values (coming e.g. from the launchcontrol, or ini) hierarchically, i.e.: spectral.alpha.cal.learningrate. Some flexibility is also offered by renaming/recasting in postprocessing which could e.g. cast midi.note45 to control.learningrate, with both the calibration and plotcontrol module reading that one instead.
7) The undefined control signals in generatecontrol/AVmixer/sequencer, can, in fact, be considered MIDI, and output as such

If you agree, it would be good to create a to-do list (if you agree of course). So far:
1) Replace 'prefix' field in each module with appropriate generic one (of add if not present).
2) Implement and check hierarchy in output of each module, e.g. in plotsignal, plotcontrol, smoothing and calibration.
3) Add calibration input values (not keys) in a new field in the output (consistent naming with .ini)
4) Make plotcontrol and plotsignal extract the right parameters entered by the calibration module 
5) Reduce different midi input/output modules to the single inputmidi and outputmidi, perhaps by including different .ini files?

robertoostenveld, 2017-11-28:
Thanks for the broad list of ideas. We might want to separate some of them to other issues to keep the discussion more clear. Let me give some first comments. In general 

Re 1 and the section before. I like it, but realize that magnitude (temperature) and unit (degrees Celsius or Fahrenheit) are convoluted. E.g. for the launchcontrol sliders the magnitude would be "position", the unit would be "midi" (not according to https://en.wikipedia.org/wiki/International_System_of_Units, but I think we can live with a slightly wider definition of units). If we calibrate the EEG alpha power, we are changing the unit (from uV^2 to midi) but not the magnitude. 

Re 2. I consider that there are three things represented in the name: the magnitude, the processing step and the unit. E.g. alpha.smooth.midi. We have separate modules that estimate some magnitude (heartbeat), some that change the magnitude (smooth), and some that change the unit (from inter-beat-interval to bpm).

Re 3. agreed, although I am not sure about pre/postfixing and/or renaming the middle.

Re 4. I would also prefer less midi interfaces, but realize there is a good reason we have so many. The output to midi devices (the volca's and avmixer) is very different from the input. And for the input in launchtcontrol there is also some output (for the toggle state of buttons). If this needs to be changed (which I don't think so), we should break this out to a separate issue.

Re 5. I would say that presently outputosc (and inputosc, which already exists) are doing the mapping between two possibly incompatible namespaces. Mapping here is renaming. If we can make the names more consistent, less renaming is needed. If you compare the two implementations (input/outputosc) the big difference is that on the output side there is recalibration (=ugly).

Re 6. I cannot follow you here.

Re 7. See item 4: I think this is more complex than simply control values. OSC does not differentiate between control/gate getset/pubsub., whereas MIDI does. And there is the mapping between name spaces to consider (e.g. consider the mapping between DNS and IP addresses, which both have some structure, but the structures don't map one-on-one).

Regarding your TODO items. I think these are more detailed than we need at this moment. I think the TODO should be: how do we want pubsub/getset items to appear in REDIS. I see the left versus right extension. Consider DNS, where data.donders.ru.nl goes from right (top) to left (detail), whereas URLs like /workshop/natmeg/preproc goes from left to right. OSC is clearly left-to-right. Thinking at an abstract level, you could compare this to RESTfull web services, which also have an identification process. E.g. http://localhost/index.html (protocol, host, resource) could be compared to redis://localhost/lanchcontrol/control077/midi (protocol, host, resource, detail). Some modules add resources (e.g. launchcontrol, spectral), and some only change the existing resources (calibration).

stephenwhitmarsh, 2017-11-28:
Thanks for reviewing the points
1) I agree we could do with a loose definition for now. I think we want to go in this direction. But regarding what you say here about calibration, we could consider calibrating to the same units, i.e. midi to midi, and uV^2 to uV^2. This would make things more traceable and require changing units only in postprocessing, only if needed. Modules should be able to identify the units of their input and act accordingly, without needing to change units in redis with postprocessing. 
2) The problem with that order of things, is that its unclear to me how we would deal with multiple output from a single process, i.e. i imagine the calibration outputting several parameters. Furthermore, if we stay in the same unit, starting with the unit in the key, will make it easier for modules to identify them (i shudder thinking that we would need to scan the fields for hard-coded unit names). Unless you think we can always _end_ the key with the unit, moving it up after each processing step (kinda ugly as well).
3) Okay, let's talk about it, it might not be clear why we thinks what we think about this :-)
4) Agreed, I did not think of the bidirectionality of the launchcontrol. No big issue.
5) Indeed, downstream (e.g. received in pure data), it is easy to handle units if they are described in the message in a standard place (i.e. the first field).
6) This refers to #122, to which I suggest we should not yet implement this before we solve the namespace issue so we don't use it as a workaround.
7 and later) I don't follow you here. But if you mean we should define the namespace, then yes, that is the first priority so that we can really nail down how we will implement the calibration which will be heavy on redis. 

Let me know when you can Skype, because I think this is about as far as we can work it out here.

robertoostenveld, 2017-11-28:
prior to fixing "units" we might want to look at the [levels of measurement](http://psc.dss.ucdavis.edu/sommerb/sommerdemo/scaling/levels.htm). Most of our values are ratio values, where an expression like "2x as large" makes sense. But a challenge is that some are bounded (e.g. midi), and some are not (e.g. alpha power). The bounded ones can be mapped onto each other e.g. 0-127 -> 0-1. For some unbounded ones we can make a standard mapping based on the distribution (e.g. Gaussian distributed variables can be normalized). But for some we don't even know the distribution...

stephenwhitmarsh, 2017-11-28:
Yes, and just to throw this in: at least so far, the electrophysiological measures are unbounded (if one disregard the bitdepth of the AD) to begin with, while the others, which for all intends and purposes can be considered control signals, could be (forced/expected to be) bounded. Furthermore, there seems is a close correspondence between the categories of units and their levels of measurement. The problem with the unbounded measures is exactly the reason we need calibration. To me that is the whole point - calibrate them dynamically to get them between bounds and allow their mapping to control signals.

robertoostenveld, 2017-12-27:
@stephenwhitmarsh and I talked things over in a Skype call and agreed  that we would initially only work on using more consistent values in redis. All bounded values (e.g. MIDI) should be represented as float and scaled between 0 and 1. 

The names of the keys should not change. When a module reads a bounded control signal  from redis, it should assume by default that it is between 0 and 1 and should scale it to its own purpose.

closed 2017-12-28 by robertoostenveld