eegsynth/eegsynth #128: record module should support wav format as output format

robertoostenveld, 2017-12-05:
I suggest to add the option

```
[recording]
format=edf or wav
```

This is useful for the output of cogito. It will probably be limited to a small (2?) number of channels.

robertoostenveld, 2017-12-05:
```
f = wave.open(file, 'w')
f.setnchannels
f.setsampwidth
f.setframerate
f.setnframes // we don't know how long the recording will be
f.writeframes // this also updates the header
f.close()
```

robertoostenveld, 2017-12-05:
as a note to myself and @stephenwhitmarsh: I could also make an adapter cable that converts line-out to microphone-in. This would allow the audio output of the hifiberry-pi to be connected to a phone. 

The zoom H2n can be used directly for recording: it has a "line-in" which is a stereo ⅛" mini phone jack that can accept two channels of mic- and/or line-level signals

stephenwhitmarsh, 2017-12-05:
Re. Zoom, true. It can also function in monitor mode, i.e. record and send
at same time to e.g. mixer

commit d6e5334a (2017-12-16): FIX #128 - implemented writing of continuous values from fieldtrip buffer to to wav file
closed 2017-12-16 by robertoostenveld