eegsynth/eegsynth #169: synthesizer module expects MIDI range rather than 0-1, suggesting generic solution

stephenwhitmarsh, 2017-12-29:
specifically in VCO_pitch and LFO_frequency.

Mapping 0-1 to pitch/frequency (Hz), o MIDI to Hz is documented here (although focused on another integer representation): https://en.wikipedia.org/wiki/MIDI_tuning_standard

"The frequency range starts at MIDI note 0, C = 8.1758 Hz, and extends above MIDI note 127, G = 12543.875 Hz"

Conversion could therefore be something like: 
Redis_value = MIDI_value/127
frequency = 8.1758 + Redis_value * (12543.875 - 8.1758)

I suggest we make a function for these conversions:
[Redis_value] = MIDI2Redis(MIDI_value)
[frequency] = Redis2Freq(Redis_value)
[MIDI_value] = Redis2MIDI(Redis_value)

stephenwhitmarsh, 2017-12-29:
Applies to Volco modules as well

robertoostenveld, 2017-12-29:
volca modules have already been updated and tested, so I don't think that those need further change.

robertoostenveld, 2017-12-29:
There is now
            FREQUENCY = math.pow(2, (vco_pitch/12-4))*261.63
which is the similar to what is mentioned on the wikipedia page. 

The only think that I see as issue is that vco_pitch is not between 0-127 any more, but between 0-1.

robertoostenveld, 2017-12-29:
I have the midi piano next to me and could use that to check/calibrate the software synthesizer.

robertoostenveld, 2017-12-29:
I have just added scaling from Redis to MIDI for the VCO_pitch control signal. I have not changed the LFO_frequency yet. I think that the whole note-sending needs some more thought.

stephenwhitmarsh, 2018-02-05:
I think this is implemented across the board now. We have decided that, when conversion is used the offset and scale remains an option in the ini. @robertoostenveld - do you agree we close this now?

robertoostenveld, 2018-02-05:
agreed

closed 2018-02-05 by robertoostenveld