eegsynth/eegsynth #170: sequencer module expects categorical input

stephenwhitmarsh, 2017-12-29:
it chooses one of several input patterns. The number of patterns is arbitrary.

One solution is to make boundaries for each pattern by something like 1/(nr_of_patterns), and select a pattern by the closest value to the 0-1 Redis value. 

This might be done by creating a function:
category = CV2Category(Redis_value,low,high)

robertoostenveld, 2017-12-29:
the software sequencer writes a sequence of control values in Redis. These used to be scaled according to MIDI and bounded between 0 and 127.  Should those control values not simply be scaled between 0 and 1?

The values could remain integers between 0 and 127 to keep their interpretation as MIDI notes easy, and there could be an additional section

```
[output]
scale=0.00787401574803149606
offset=0 
``` 

PS note that Redis does not distinguish between integers and floats.

stephenwhitmarsh, 2017-12-29:
Keeping MIDI values + scaling sound like a good plan.
What are your thoughts on selecting the pattern?

robertoostenveld, 2017-12-29:
so far it was done with a knob of the launchcontrol.

e.g. launhcontrol.control021 would return 0-127, which after quantization would become 1,2,3,4

commit eda636cd (2017-12-29): FIX #170 - implemented output scaling for the sequencer module
I also added more sequences to the ini file, and improved the debugging output
closed 2017-12-29 by robertoostenveld