eegsynth/eegsynth #172: unclear meaning of 'physical min/max' in recordcontrol

stephenwhitmarsh, 2017-12-29:
While i suspect there will be no issues in the recordcontrol module, which writes control values to .edf, the exact mapping from control to edf values is not documented, although the code suggests a physical max/min range of +/-1024 sampled by 16 (?) bit values o +/- 32768. Dividing the two suggests that for floating point values between 0-1 only 32 values are available. I might be wrong, but it would be good to match record and playback, and optimize them to have enough resolution for 0-1.
Ofcourse, the issue will be clipping. We should therefor make a decision for the 'physical' max/min range. Something like -10:10?

robertoostenveld, 2017-12-29:
the physical min/max reflect the range of bounded and unbounded signals in Redis (or FT buffer). 

For bounded we have decided that it will be between 0 and 1.

For unbounded variables I think that the range is up to the normalization/calibration module. But unbounded values will remain unbounded after scaling, unless also clipped.

I like the idea of ensuring that the signals are clipped to the physical min/max values prior to being written to the EDF file. When not clipped, weird things can happen.

stephenwhitmarsh, 2017-12-29:
Okay, for physical min/max shall we put the default in the code to 0-1, but also it to be set in the .ini?
And indeed implement clipping outside that range.

closed 2018-01-07 by stephenwhitmarsh