eegsynth/eegsynth #173: rethink quantizer module

stephenwhitmarsh, 2017-12-29:
It is unclear what the quantizer module expects as input and how it maps to the quantized scales. Partly this is due to lack of documentation, but also perhaps a bit of an idiosyncratic implementation.
I suggest we rethink the module and make it consistent with (extended) functionality of quatizers in synthesizers:

[INPUT]
1) control value to be quantized
2) attenuation/gain of (1) (chromatic range before quantization)
3) offset of (1) (chromatic transposition before quantization)
4) scale (iodic, lydic, chromatic, etc)
5) offset after quantization (transposition in quantized scale, 0-1 will transpose within one octave)*

[OUTPUT]
6) quantized control value
7) quantized control value + transposition (5)

*) nice modules such as the microscale by Intellijel also allow setting the transposition in (5) to 'within scale', 'same key', 'octave' and the like. But that might be too much for our purposes now :-)

robertoostenveld, 2017-12-29:
the code is only some ~15 lines and maps each control signal value (of the selected channels) to the nearest level and returns the specified quantization value.

So if 
```
level=1,2,3,4
quantized=1,2,3,4
```
then a value of 1.7 is closest to level 2 and therefore is output as 2.

And if
```
level=1,2,3,4
quantized=42,65,16,27
```
then a value of 1.7 is closest to level 2 and therefore is output as 65.

The confusion might be that it is done for multiple input channels and that multiple quantization outputs can done at the same time.

robertoostenveld, 2017-12-29:
we discussed this over Skype, the conclusion is that the implementation does not have to change at the moment. There should however be scaling and offset support for input and output. Furthermore, the example in the ini should be improved.

commit 76a86c6f (2017-12-29): FIX #173 - implemented scaling of input and output values, improve feedback, improve documentation and example in ini file
commit 8c4dfd35 (2017-12-29): Merge pull request #176 from eegsynth/issue173

FIX #173 - implemented scaling of input and output values, improve fe…
closed 2017-12-29 by robertoostenveld