eegsynth/eegsynth #182: document the sequencer and quantizer...

robertoostenveld, 2018-01-07:
as these seem to be a bit hard to understand right now.

What I have in mind is to make a short markdown file with a specific ini for both, combine that with the (software or hardware) synthesizer or and then make a movie for youtube.

To control the sequencer I plan to use 16 knobs (pitch) and 16 buttons (on/off) from the launchcontrol. Those create control signals in REDIS, which I want to map onto a nice harmonic scale with the quantizer. The output of this goes to the synth.

robertoostenveld, 2018-01-08:
it would be nice if the quantizer would be more flexible in the input-output mapping, i.e. if the input list and output list would not have to be the same length.

It would be nice if the different notes on the keyboard can be triggered  by a single pubsub event (where the value is the pitch).

it would be nice if the sequencer would allow values from redis as pitch, at the moment it only accepts hard-coded values in the ini.

robertoostenveld, 2018-01-08:
oops, I should have pressed return on the comment prior to pushing the branch

stephenwhitmarsh, 2019-01-06:
Concerning the documentation - the idea sounds good. To be clear this would also include the generateclock module.

I would prefer the first write the markdown file as self-explanatory though. 

Concerning videos, I have just inherited a very nice handycam and plan to design some videos as well, but plan to think about it together, covering the basics first. A video on generateclock, sequencer and quantizer would not be top priority.