eegsynth/eegsynth #186: Controlling current for EEGball game

stephenwhitmarsh, 2018-01-08:
Charles Findling from the brain control club started making a ventilator+tube system to be controlled by the eegsynth. I worked it out to a rather cool working ventilator+ball system:

https://www.youtube.com/edit?o=U&video_id=IO3PKoYIYO8

Alas, the planned way of controlling the current to the fan did not work (the electronics could not handle 12v). Also, the digital part of the solution might be a bit overkill: Charles uses an Arduino with a 'Linux Shield', on which Redis and a python script runs, interfacing with the Arduino code, which controls a PWM pin. More consistent with the EEGsynth would be to either 
1) to create an Arduino-output module + Arduino code
2) run the whole thing on a Raspberry Pi

I propose to discuss a (general) way to control currents in such setups (or e.g. just an analogue lamp (gloeilamp)), with a minimum of electronics. I suspect there will be plenty of documentation online starting with an Arduino. Ofcourse, we have all the necessary code/schematic for the CVgate device (@robertoostenveld: did you correct the schematics?)

I will ask Charles to share/create his github account so he can participate in the discussion.

robertoostenveld, 2018-01-08:
The video is (yet) not accessible on youtube, can you fix that?

I suspect the fan is a based on a simple DC motor, which means PWM? For example like [this](http://www.electroschematics.com/9540/arduino-fan-speed-controlled-temperature/). That does not require a full linux stack to run, just an Arduino and a transistor to handle the 12V. And possibly (as in the link) an RC circuit as low-pass filter.

If it is indeed PWM controlled, the easiest would be to re-use the [arduino code](https://github.com/robertoostenveld/arduino/tree/master/eegsynth_cvgate_mcp4725) for the 1-channel CV/Gate interface. That would allow to re-use the cvgate module. If you would change the code such that the input values are divided by 16, then the 12-bit value of the ADC maps onto the 8-bit  value of the Arduino PWM and we don't have to do any additional scaling.  The serial protocol of the eegsynth_cvgate_mcp4725 firmware now allows for 4 channels. You could e.g. use that for multiple fans (multiple people playing together?).

robertoostenveld, 2018-01-08:
Btw, the schematics for the 4-channel CV/Gate is still not correct, pending issue #123. But you don't need that schematics (or that of the 1-channel version), just the piece of the firmware for the communication over serial USB port.

stephenwhitmarsh, 2018-01-08:
From Charles:
---
Hey,

Looked at it, found this transistor which works find, https://www.arduino.cc/documents/datasheets/MosfetTransistor.pdf. I actually tested that one already at the maker lab!

I gonna buy one, it is only a few euros!

stephenwhitmarsh, 2018-01-08:
video made public: https://youtu.be/IO3PKoYIYO8

robertoostenveld, 2018-01-08:
thanks, impressive tube! It seems to have a quite smooth control. 

I don't know what kind of fan you are using, but my guess is that you would not need a power mosfet. The one in the data sheet is rated for 100V and 10A and for a max power dissipation of 50W. You'll need to point the fan on the transistor to keep it cool ;-)

A low voltage transistor should work as well (and is only a few cent).

stephenwhitmarsh, 2018-01-08:
http://blog.ninapaley.com/2018/01/01/24-free-goddess-gifs/

It's a 12V computer PS fan, of 300mA if I remember correctly. 
Nice link to the arduino fan! 

I'm opening up to picking up the soldering iron and Arduino again...

robertoostenveld, 2018-01-08:
if you search for "fan arduino" you'll find many examples.

robertoostenveld, 2020-04-13:
This is not something that the EEGsynth software repository will address

closed 2020-04-13 by robertoostenveld