eegsynth/eegsynth #188: there is no postprocessing of the value in pubsub triggers

robertoostenveld, 2018-01-08:
Triggers are implemented using Redis pubsub and also have a key and value. The key is for example launchcontrol.note073 and the value is the velocity of depression. For the Launchcontrol that velocity is always 127, but for a keyboard it can be between 1-127. The value 0 indicates that the key has been released. 

The specific key of a keyboard can also be coded in the trigger value (as in launchcontrol.note, which holds 73 as value). In that case there is no possibility of representing the velocity, or the release of the key.

At this moment we do not have any postprocessing of the trigger value. Furthermore, I a not 100% sure whether the trigger value is always treated consistently with a scale and offset.

robertoostenveld, 2018-11-21:
considering the naming of the modules, we might want to consider
* processsignal (now preprocessing)
* processcontrol  (now postprocessing)
* processtrigger

stephenwhitmarsh, 2018-11-21:
Some idea's on naming:
'preprocessing' is a term well-known to EEG community, and the
functionality of the module is intended to be consistent with what is
expected there, while only 'process' is rather non-informative.
I agree on the arbitrariness of the pre- and post-processing modules, the
false chronology of pre/post, and see where you are coming from.
I think we might actually just be more precise in the functionality of the
modules:
- 'preprocessing' --> 'filtersignal' (or 'signalfiltering'), and
rereference (or 'signalrereferencing'), althought the latter would probably
be better implemented in the e.g. the OpenBCI module as I do not (yet) see
the use of rereferencing online in patch.
- 'postprocessing' --> 'controlfunction', 'evaluatecontrol' (my favorite),
or 'controleval'. Also, we should document the compressor, as well as other
EEGsynth library functions separately, not in the 'postprocessing' module

- 'processtrigger' --> unclear yet what the functionality should be (which
is a good thing!)?
- It brings up the point that we (implicitly) treat gates and triggers as
control signals, which still makes a lot of sense to me - they should/could
be treated as binary control signals (between 0 or 1, but represented as
floats with certain precision?, i.e. with 0.5 as threshold either way?),
and be 'patched' in a similar way? Will that work or break the code? Should
we represent them as binary variable types, or can we deal with floats
here? It does mean that we then scale the MIDI velocity values (/255) to
0:1. Think out loud, that means that 'real' velocity cannot be encoded.
This might be a blessing because it enforces a clear idea of gates/triggers
and binary values/events. However, perhaps we shouldn't throw away the baby
with the bathwater, and just scale, hoping we won't run into precision
issues.
- Just to be clear, I don't think it's a good idea to use 'processtrigger'
to do the scaling on the level of the 'input', scaling e.g. velocity to 0-1
should be done by the input modules, e.g. Launchcontrol, I'm sure you agree.
- I don't know what functionality you had in mind with 'processtrigger',
but let's put some ideas together and see if something consistent comes
out, e.g.:

1) triggers-to-gates, e.g. by making them flip-flip a 'gate' (control value
between 0 and 1)

...oops your email just crossed me. Seems I should keep it at this now.

robertoostenveld, 2018-11-26:
The processing could relate to the value of the trigger, but also to the timing (and relative timing) of triggers. E.g. the rate of a trigger could be estimated, or the regularity, or the change in rate (speeding up or slowing down).

robertoostenveld, 2019-09-01:
I have implemented a module for this. The ini file looks like this:

```
[general]
delay=0.05
debug=2

[redis]
hostname=localhost
port=6379

[initial]
; here you can specify the initial values of some control values
launchcontrol.note041.counter=0
launchcontrol.note042.counter=0

[trigger]
t1=launchcontrol.note041
t2=launchcontrol.note042

[input]
; the keys here can have an arbitrary name, but should map those in the output section
; the keys must be lower-case. values should not contain an equation, only one-to-one mappings
x1=launchcontrol.note041.counter
x2=launchcontrol.note042.counter
x3=launchcontrol.control077
x4=launchcontrol.control078

; for each trigger one or multiple computations have to be specified
; besides +, -, /, *, the equations also support log, log2, log10, exp, power, rand, randn from numpy
; and compress, limit, rescale, normalizerange, normalizestandard from EEGsynth

[t1]
launchcontrol.note041.counter=x1+1    ; increment with one

[t2]
launchcontrol.note042.counter=x2+x3   ; increment with the value of a slider

[output]
; after completion of the computation, the trigger is resent with this prefix
prefix=processed

```

The functionality is mostly the same as `postprocessing`, except that rather than doing the computation every 50ms, the computation is done upon receiving a trigger. It listens to multiple triggers, and computes multiple sets of equations. 

When done, a prefixed copy of the input trigger is sent, which also allows for chaining triggers.

commit 39572123 (2019-09-01): FIX #188 - implemented processtrigger module, largely copied from postprocessing
closed 2019-09-01 by robertoostenveld