eegsynth/eegsynth #197: synchronization module is outdated

robertoostenveld, 2018-01-11:
although it does scaling on the input value (from redis), the naming in the ini is not consistent with other modules.

It now has 

```
[input]
rate=60         ;launchcontrol.control077, heartrate.channel1
offset=64       ;launchcontrol.control049
multiplier=64   ;launchcontrol.control077 64=center, 0.25,0.33,0.50,1,2,3,4 four steps up and down
```

In the code I see hard-coded values that relate to mapping between Redis and MIDI.

robertoostenveld, 2018-01-12:
I just realized that the multiplier is slightly different than the normal scale/offset option, as it also serves to do integer-wise division/multiplication of the rate.

robertoostenveld, 2018-01-12:
rather than having a continuous control signal for the integer rate division/multiplication, it would also be possible to send multiple Redis events. For example like this

synchronization.heart     ; this beats at the original rate, e.g. 70 BPM
synchronization.heart.times2 ; times 2, e.g. 140 BPM
synchronization.heart.times4
synchronization.heart.times8
synchronization.heart.div2 ; divided by 2, e.g. 35 BPM
synchronization.heart.div4
synchronization.heart.div8

But this does not work for the MIDI and the Serial output, which are needed for analog CV/Gate. Even then, I wonder whether the multiplication should be done inside this function, or using the postprocessing module, which could be used to multiply the rate control signal. The latter would be more consistent in the recent view of not replicating the calibration and scaling of signals in each module.

robertoostenveld, 2018-01-12:
@stephenwhitmarsh, have you ever tested the synchronization module (MIDI out) with the gates of the  Endorphines Shuttle Control? I know it works with the Kork Volca's to drive their internal sequencer. It uses the "clock" MIDI message.

stephenwhitmarsh, 2018-01-17:
Some functionality of the endorphines shuttle control can be used with gates on specific or any note, on specific channels, such as clocked LFOs. 
In the latest firmware also has synchronization functionality based on the MIDI clock signal. From https://www.endorphin.es/cargo3/ :
---
Syncronization | Start/stop gate returns +5V when the playback on the DAW is on, and 0V when is stopped. Separate 1ms trigger outputs from DAW remore controls (play, stop, continue). CC48 reserved for the shuffle of the MIDI clock incl. all dividers but excluding PPQN*2. CC48 value 0 corresponds to zero shuffle, CC48 value 127 corresponds to maximal shuffleRandom CV generators (from v.3.3) that are generated under incoming MIDI clock. Start/continue enables them to generate and Stop makes them pause. CC#12 is reserved for choosing the divider for incoming MIDI clock (1/32 (PPQN/3), 1/16 (PPQN/6), 1/8 (PPQN/12), and 1/4 (PPQN/24)). CC#13 is reserved for random CV distribution (for quantized and stored voltage generators)
-- | --
Tapped clock | Three or more taps is enough to define the frequency. CC49 reserved for the shuffle of that type of clock. CC49 value 0 corresponds to zero shuffle, CC49 value 127 corresponds to maximal shuffle
----

robertoostenveld, 2018-01-17:
nou nou, wat ben je aan het schreeuwen ;-)

Het is me nog niet duidelijk hoe het werkt. Heeft jou endorphines de laatste firmware?

stephenwhitmarsh, 2018-01-19:
Ah, yeah, sorry about that :-)

First I think it's good to get an overview of the different ways in which synchronization is accomplished within and between music equipment, using MIDI and CV/gate/triggers (both of which the Endorphines Shuttle Control straddles, especially with the latest firmware I installed).

# In MIDI there are two main ways, either using [time codes](https://en.wikipedia.org/wiki/MIDI_timecode) similar as in video equipment (not relevant for us) and clock signal. 

# the MIDI clock signal consists of a rate of pulses. The speed of the pulses sets the clock speed of sequencers, drum computers or a DAW, via 24 Pulses Per Quarter Note (PPQN).  This speed can be considered the 'base-rate', which (often) is divided (clock-divider).

# In the analogue domain, triggers/pulses can used (short duration +5v of several ms, in contrast to longer (and variable) gate signals) to set a the clock speed. However, rather than the previous rate-encoding, a single trigger will code for a single (quarter) note, or step. The [Doepfer Darktime](http://www.doepfer.de/pdf/Dark_Time_Manual.pdf), e.g., can be used as a slave to  a USB clock (24 PPQN), and output analogue triggers (one per step). In such patches a single pulse triggers an event or advances to the next note in a sequencer. 

# In analogue modular setups synchronization between modules is often done more flexibly, using modules that respond to 'taps' or 'pings'. These are gate or triggers/pulses that the module then uses to extract a rate. Three or more of those 'taps' are needed to set the speed. The speed can then be locked or kept adaptive. In fact, they often adapt their rate (synchronize) a bit slowly, which is a rather prized musical characteristic. Good examples are the [4ms pingable LFO](http://www.4mspedals.com/qplfo.php) and [clock distributor](http://www.4mspedals.com/qcd.php) with added divider/multiplier.

* MIDI synchronization between MIDI and modular devices seems to be a general pain, with latencies, jitter and drifts being very common except in perhaps the most sophisticated professional studio equipment. 
* The Shuttle Control can deal with both MIDI clock signal, and adds a divider/multiplier as well as respond to taps with a divider/multiplier. 
* I think it is important to consider that MIDI clock signals are designed to be used as a base-rate, i.e. a stable speed and synchronization of events between devices (sequencer and drum machine e.g.). It is not designed to code for smoothly or quickly changing rates. For those it makes more sense to use gates to tap/ping a more loose but responsive system.

closed 2018-01-20 by robertoostenveld

robertoostenveld, 2018-01-23:
the input Redis channel that controls the "rate" option (called like this in the ini) is - after rescaling with scale_rate and scale_offset - the beats per minute, i.e. its units are "1/minute" (or events-per-time). The beats correspond to the quarter notes. 

The PPQN is not expressed in absolute time, but in pulses per beat, i.e. it is an integer without units. 

I have implemented in the code an internal clock thread that uses 24 threading.Event() objects to allow the other three threads (midi, redis, serial) to remain linked to the clock-thread. The midi thread uses all 24 PPQN to send a clock signal. For the redis and serial thread you can specify the PPQN, although it must align with the 24 PPQN.

This corresponds to the "step" option in the ini. This can directly be interpreted as PPQN and allowed values are 1, 2, 3, 4, 6, 8, 12, 24. 

To synchronize the Volca beats (connected with analog CV over usb) and the Volca bass (connected over MIDI over usb), I had to configure steps=2, i.e. the Volca uses 2 analog pulses to increment one beat. With MIDI at 24 PPQN (hardcoded) and analog triggers at 2 PPQN (steps=2), the two Volca's remained perfectly in sync over a wide range of BPMs.  The analog triggers PPQN is something that I actually can measure on the Volca sync output with a scope... I will test this.

Oh, and note that I have also implemented an "adjust" option, next to the "steps" option. You can shift the output trigger to occur at different internal steps. So with steps=4 and adjust=0 you would have (zero-offset, with PPQN going from 0 to 23)

````
0     6     12     18     
````

and with adjust=2 you would get

````
2     8     14     24     
````

This allowed me to align the bass and drums in a nice fashion (on-beat, or just off-beat). I better demonstrate all of this with a live Skype connection...

robertoostenveld, 2018-01-23:
PS I think that the threading might still work with 96 PPQN, where the MIDI clock message would be sent on every 4th, but I doubt whether 960 works in Python running on a general purpose multi-tasking PC. With 120bpm, or 2 beats-per-second, that is 500ms per quarter note, which is ~0.5 ms per pulse. 

But then, it is something we can try out.