eegsynth/eegsynth #202: implement synchronization with endorphines

robertoostenveld, 2018-01-20:
see issue #197 and the valuable documentation there

stephenwhitmarsh, 2018-12-12:
Using the clock signal works wonderfully with the Endorphines Shuttle Control. However, it needs a MIDI start message to start. This is similar to the Doepfer Dark Time, but there I could just press the start button. For now, I just added in line 214 of generateclock.py (not committed):

            # MIDI start message
            msg = mido.Message('start')
            midiport.send(msg)

There are several non-exclusive options for a better implementation:
1) define (launchcontrol) event/pushbutton for MIDI-start and one for MIDI-stop. Could also be implemented in the outputmidi module
2) MIDI-start at start of module, MIDI-stop at closing (keyboard interrupt)

Suggestion: Add a start and stop value in the [midi] field. If boolean just start/stop automatically, else respond to redis pubsub event. Also add to outputmidi.

robertoostenveld, 2018-12-12:
e.g. 

```
[midi]
device=xxx
ppqn=24
play=launchcontrol.note021  ; instead of the general "midi" item
```

upon play it sends "start", when play goes to 0 it sends "stop"

stephenwhitmarsh, 2019-06-14:
If I remember corrected, this is now implemented, right?

robertoostenveld, 2019-06-14:
this does not seem to be implemented, it is not in `endorphines.py`