eegsynth/eegsynth #208: Signal2control module ideas

stephenwhitmarsh, 2018-02-24:
After exploring electrogastography, it became clear that this poses some (expected) practical problems.
EGG signal is at around 0.05 Hz, requiring minimally a minute long timewindow for the power estimate. More problematic, is that the fluctuations of power are expected to occur even slower. Which makes automatic/manual calibration very tricky indeed. I see two solutions:
1) further develop a manual calibration, which would include somehow visualizing manually-set max/min. This could be done in plotcontrol, but it is also clear it would need colors&labels to distinquish the different control signals in a single plot. Still, this would be somewhat regressive, as we have found manual calibration to be inconsistent with our ultimate goal of automatic processing. However, there might be future uses of this and we should just bite the bullet IMHO.
2) There are more features that are interesting in EGG. First of all the phase. I can image a preprocessing module (somewhat wide) bandpass filtering, and then sampling the instantaneous phase to create a control signal. This would then sonify the phasechanges over time, and reflect changes in frequency (speed of phase change) as well. I think it might also be more robust in terms of calibration as this does not concern a smoothed signa (i.e. not based on a sliding timewindow of amplitude), and could therefor be applied over 'just' a couple of minutes. This would require a new module that, according to a set 'stepsize' or 'delay' take the last sample in a buffer, and put that in redis as a control signal. Additionally, one could use an average over a certain timewindow. Alternatively, one could base the sampling on the timestamp, but this requires a different use of the FTbuffer as we have done so far. What are your thoughts on this @robertoostenveld?

Cheers,
Stephen

stephenwhitmarsh, 2018-02-24:
Another use of the second tactic would be for the accelerometers and motion capture signals.

robertoostenveld, 2018-02-24:
considering Fourier analysis: effects of spectral leakage will be very large for the 1st approach, since the window cannot be made arbitrary long. When you make it too long the effect of control will disappear.

I think that temporal analysis (with appropriate filters that have very long filter kernels) will be better. But estimating phase will be hard if the signal is not very periodic (1) and if its amplitude fluctuates (2). Both 1 and 2 cause the signal to become wider in the fourier spectrum. 

I propose to start developing something in MATLAB, ignoring the noise for the moment.

stephenwhitmarsh, 2018-02-24:
Thanks. I do want to note that the EGG is, in fact, very periodical
(forgetting noise for a moment), with phase estimation arguably quite
effective in my previous lab. I am concerned about the neccecary filter
order however...

robertoostenveld, 2018-02-24:
With which precision is it periodic? You write 0.05 Hz, which given rounding can be anywhere between 0.045 and 0.055, i.e. 10% uncertainty in both directions. 

Alpha tends to be periodic with some 5% frequency jitter (half a Hz both directions), already causing a significant widening of the peak if you look at the spectrum. Theta has a wider frequency spread than alpha. Gamma can have 20-30% frequency spread.  

If there is an amplitude modulation at M Hz then the spectral peak (relevant for estimating phase) will be plus-minus M Hz wider. If the AM is not at a single frequency, but a mix, the sidebands will be wider. All logic for “normal” frequencies applies to 0.05Hz signals as well. E.g. I would only do a wavelet analysis with at least three cycles (which is 60 seconds here) or more.

stephenwhitmarsh, 2018-02-26:
Yes, ofcourse, I agree with the width of the EGG spectral peak. What I wanted to emphasize is that in my previous lab we have been arguably successful in identifying the EGG peak and extracting the phase. The example picture here is ofcourse of a representative subject - it is different from subject to subject: 
http://iec-lnc.ens.fr/IMG/Files/Visual/Publications/2017_Richter.NIMG.pdf

But you are absolutely right that with a relatively small filterorder (still huge given the low frequency), we will have a pretty wide power/phase estimate. For the analysis of variance in the phase, the width of the filter allowed shifts in frequency. If I remember correctly, something like a half-width of 0.01 Hz was used. Practically, however, the filter-order was really a matter of a couple of minutes. Still, the point I wanted to make is 
1) that we do see to find a peak in the spectrum around EGG freq (0.05Hz), a bit higher in this subject: https://www.youtube.com/watch?v=HyRIeE0ZnX8
2) using a phase-estimate of a bp-filtered signal will give us a more interesting control signal, which I agree will be under the influence by changes in amplitude. In fact, in my last experimental data, strange shifts in phase were caused by drops in signal (amplitude). However, it is a matter of 'doing the best we can'. How good that is, we should just have to see.

I agree we should do some offline tests. I will measure some EGG, say 20 minutes, this/next week, and take a look.