eegsynth/eegsynth #209: Les Beaux Arts COGITO workshop

stephenwhitmarsh, 2018-02-26:
Together with Daniela de Paulis, Vincent Rioux and his students, we will create a workshop/installation. One idea that has been suggested is spatial sonification in accordance with spatial brain activity. Below I posted my considerations, insisting in using a similar approach as we have been taking before. It creates an interesting question on how to best calibrate over multiple sensors, in a way that retains their differences / topography. The most basic approach would be to calibrate all channels by the max and min of the max and min per channel. Turning this around in head it seems that would actually be enough, and robust.
 
Implementing this with the postprocessing modules seems a bit tricky though, or at least will require an unreasonably long syntax. In fact, I have encountered this in our previous performance as well. Things would be very much easier is we would either allow/implement the [input] declaration in the .ini to contain syntax evaluation as well (i.e. max/min/+/- etc). @robertoostenveld - what would you say?

====

Sonifying 'raw' EEG signal would not be the way to go, for practical and theoretical reasons: 
a) we have quite an issue with aliasing/calibration in the current implementation, introducing artifactual rhythms, which we have not been able to cure, and not sure we will
b) the COGITO implementation is inconsistent with the general framework of the EEGsynth, and bypasses any 'analysis', i.e. identifying and filtering for alpha activity, which would be the interesting element. 
c) sonified raw data does not result in a signal that can be interpreted in any meaningful way. In fact, one would be mostly 'listenting' to noise from sensors/muscles/artefacts, and the exact way of sonification would be arbitrary and not easily adjusted 'artistically'.
d) implementing the COGITO module for single EEG channels would bring us back to the drawing board - the cool thing about COGITO is that it is a purpose-build protocol for sending multiple EEG channels into a single mono channel. The opposite, of doing this for each channel separately.
e) I do not see how the student could engage in this implementation as it's too specific, inflexible (concerning signal conversion), and too involved with coding.
e) doing this would not add to the EEGsynth, and so not a good time investment for me. In short, I do not want to spend time on doing it that way. 

Instead, I would approach it like this:
a) identify user-specific alpha frequency (8-14Hz) visually, and flexibly in real-time
b) analyze that alpha activity (power estimate) for each channel in real-time.
c) dynamically and automatically calibrate alpha estimates for each channel (already implemented) or over channels (i.e. maintaining differences in amplitude between electrodes, but I have to think how this would make sense and how).
d) convert them into control signals (32 control signals, or averages over several 'brain regions')
e) do all of the above on my / another Linux laptop/desktop, and broadcast those control signals as OSC over a network.
f) do whatever you want with those OSC signals using Raspberri-Pi's or whatever, to control aspects of sound using pure-data / max-MSP / super-collider
I would know how to do (a) to (e), and pretty confident it will work. (f) is where the students and you can really get your hands dirty, be flexible and explorative, without having to spend too much time making code that will never be used afterwards. 

Also note, that we could reduce the number of electrodes. One could argue that 32 channels is overkill, and not a theoretical number. Less could be more.

stephenwhitmarsh, 2018-03-06:
If that option would work, it would be great, and sufficient!

robertoostenveld, 2018-05-13:
from https://github.com/eegsynth/eegsynth/blob/master/patches/beauxarts/postprocessing.ini it seems that the Beaux Arts performance is history by now. So I guess nothing remains to be done here.

closed 2018-05-13 by robertoostenveld