eegsynth/eegsynth #244: Problem starting the data buffer

mercuriaretrograda, 2018-10-10:
Hello!
I am installing the eegsynth but in the step where you are supposed to start the data buffer (point 2 of the Starting Data Buffer section https://braincontrolclub.miraheze.org/wiki/Tutorials) I keep getting the same error: /Users/mariana/eegsynth/module/buffer/../../bin/buffer: No such file or directory

Do you have any light on what could be causing this?

stephenwhitmarsh, 2018-10-10:
Hi Mariana,
The path seems wrong, but I don't know what you are doing. If you provide more info on what you are doing, i.e. the command you are typing, the directory you are in, where you placed the EEGsynth, it will be clearer. 
Cheers,
Stephen

stephenwhitmarsh, 2018-10-11:
Also, make sure you follow the installation instructions, including running the install script which downloads the proper FieldTrip buffer executable for you OS:
https://github.com/eegsynth/eegsynth/blob/master/doc/installation.md

mercuriaretrograda, 2018-10-11:
Thanks! I was missing some steps there.

Regarding the step 3 
"Start up the buffer module, using your own ini file: ./buffer.sh -i ../../inifiles/launchcontrol.ini."

I don't know what ini file to use, what do they mean by launchcontrol.ini?

robertoostenveld, 2018-10-11:
you may want to look into the install.sh script in bin, see https://github.com/eegsynth/eegsynth/blob/master/bin/install.sh
it fetches the external binaries (like the buffer application).

The "launchcontrol.ini" was a typo, it should be "buffer.ini", I fixed it. Note that I usually do

```mac011> cd module/buffer/
mac011> ./buffer.sh 
mac011> ls
README.md	buffer.ini	buffer.sh
mac011> ./buffer.sh 
```
to start it with the default settings (which is 2 buffers on port 1972 and 1973). That works for most cases.

robertoostenveld, 2018-10-11:
oh, please use the documentation on https://github.com/eegsynth/eegsynth rather the one on the miraheze wiki. The github version is more up to date.

closed 2018-10-15 by stephenwhitmarsh