eegsynth/eegsynth #256: decide on the name for the "sequentializer" module

robertoostenveld, 2018-11-15:
This relates to #253

The current name is confusing, also because it resembles "sequencer", which is another module (both in hardware synths, as in the EEGsynth software ecosystem). The functionality of the "sequentializer" module does not resemble that of a sequencer in any way.

What about "edgemixer"? 

I think that "mixing" (multiple signals) relates better to what this new module does, and "edges" relate to the graphical explanation. See below for a sketch in the trivial bilinear AB case, for the (intended use) tripple bilinear ABC case. 

[image]
  
It does however not convey the switching from one edge to another.

stephenwhitmarsh, 2018-11-17:
geomixer?

robertoostenveld, 2018-11-18:
👍🏻

closed 2018-11-18 by robertoostenveld