eegsynth/eegsynth #262: review whether modules have an internal state or not

robertoostenveld, 2018-11-21:
All modules with an internal state should get a start/stop/reset button, just like the playback and the generate modules.

Right now I can only think of the sequencer module...

stephenwhitmarsh, 2018-11-26:
Stop = Pause?
also: historycontrol

robertoostenveld, 2018-12-10:
we should also make them more consistent

```mac011> grep -l play= module/*/*.ini
module/playbackcontrol/playbackcontrol.ini
module/playbacksignal/playbacksignal.ini
```

```mac011> grep -l enable= module/*/*.ini
module/calibration/calibration.ini
module/historycontrol/historycontrol.ini
module/historysignal/historysignal.ini
```

generateclock has the midi and redis option which act as enable/play.

sequencer does not have them (but should).

robertoostenveld, 2020-04-13:
I think that at this moment this issue does not require further attention

closed 2020-04-13 by robertoostenveld