eegsynth/eegsynth #271: send triggers and control values more consistently

robertoostenveld, 2018-11-26:
Some modules only write triggers, others only write control values, whereas some (the newer ones) send both. I propose to consistently send triggers and control values as a pair. 

I.e. when a control value changes in Redis, that change should also be sent as pubsub message. And when a  trigger is sent as message, the value of that trigger should also be represented in a control signal.

A possibility would be to implement a generic function in lib/EEgsynth, just like the SetChannel and SetTrigger functions used by some modules. It could also support debugging (by optionally writing the change) and switching triggers/gates off after a set time (as now implemented in heartrate and outputgpio). 

````
def EEGynth.setstate(key, value, duration=0, debug=0):
    r.set(key, value) # set it as control channel
    r.publish(key, value) # send it as trigger
    if debug:
        print key, '=', value
    if duration > 0:
        # switch off after a certain amount of time
        threading.Timer(duration, EEGynth.setstate, args=[key, 0.]).start()
````

robertoostenveld, 2018-11-26:
This should be changed everywhere in the code where ```r.set``` or ```r.publish``` is now used.

stephenwhitmarsh, 2018-11-26:
Yes. I agree that this would make things consistent - by 'doing everything always'. I also see a lot of improvements in flexibility and functionality if we would follow through with this and additional functionality that would flow naturally from it, e.g. identification of change of control value (*), and more efficient plotting.

However, we should be careful about increased complexity in the code, but read on: at the moment those modules that wait for a pub do that in a separate thread, while those that that only use control signals 'get' them use simple loops. Threads and loops also deal differently with handling exceptions/interrupts, something that is also harder to expect external collaborators to contribute to. Myself included, I find that although there is a more consistency, and therefor more possibility to learn from existing modules, the threshold to contribute according to our 'system' is a bit high, especially for those that deal with pubsub (they are just harder to understand in general). Before it might not have been very relevant, or prudent, to do so, because we have been evolving quite a bit ad hoc, but I think we have arrived at a point with this proposal where it is time to document our code in a more 'meta' level, as if we would be preparing for workshops (which we should be able to do once code, philosophy and documentation are consistent and explained well).

More precisely, I suggest that we first create a proto-module (called 'protomodule', or 'dummymodule' if we want to be less pretentious). This module reads from both Redis and the FT buffer, and writes to the Redis and a second FT buffer. It should have error-handling (no Redis, no buffer, no data in buffer, keyboarrd interrupt, closing threads appropriately). It will do minimal processing, such as updating a Redis value in some iterative sense (e.g. up and down toggle value when exceeding 0 or 1, with other value continuously ramping up or down) for both Redis and FT buffer, and ofcourse using both triggers and control values. This should be well documented inline in the code, as well as identify the main units - the threads/loops, preable, etc, i.e. the general architecture. I can then also document those more deeply in the /doc. In this way, we should also have the code more 'modular' and allow external contributes to 'add your own code here'. I propose also to do this first, so what we can really wrap our head around the possibilities before we implement them, and adapt this template (hopefully minimally) along the way. I also suggest that once we made this template, we split the work of adapting all the functions between the two of us - i'd rather do this slowly and carefully so to benefit most out of it, and to keep me comfortable with the code.

What do you think?

We also need to think about when to _use_ pubsub or putget. Can we identify some heuristics in this? Does it matter?

(*) Question is whether we want to be consistent in this as well (probably): only update/pubsub when a CV value _changes_, not only updated?

robertoostenveld, 2018-11-26:
Regarding writing a protomodule (a.k.a. ft_examplefunction): I don't think it would be the most efficient in getting new contributors/users up to speed. The average number of lines of code per module is only 176, which should be quite easy to digest for our audience. Rather than an example function, I suggest we point (in the to-be-written meta documentation) to good examples for specific patterns. One consideration here is that a typical module would not have all patterns at once. If we were to make one protomodule that contains all patterns, it would be overly complex. But I do agree on meta documentation!

Re when to use one or the other: I would say that in many cases set/get and pub/sub could be used interchangeably. The set/get pattern is much simpler in the code, but also less efficient. The crucial difference is in proper-synchronized versus non-synchronized processing, where non-synchronous in our case translates in many cases to more-or-less-synchronous (polling 10 or 20 times per second). 

The idea of ```EEGynth.setstate``` as proposed above in a way similar to ```EEGsynth.patch```, where I don't want to think about whether the value is in the ini or in Redis. I guess "setstate" is totally wrong as name, rather it should be "updatevalue" (where the value is presented on a 3.5 mm output connector, or in our case in Redis).

robertoostenveld, 2018-11-26:
Now that I am looking at the EEGsynth.py code, I realize that ```updatevalue``` might be a method of the ```EEGsynth.patch``` class.  This makes it more symmetric with the ```patch.getstring```, ```patch.getint``` and ```patch.getfloat``` methods. 

The difference is that in Redis we don't have sections like we do in the ini files.

robertoostenveld, 2018-11-26:
For the handling of events we now also have a quite standard pattern, which means that we could also consider making that into a function of the EEGsynth object. That would make the code much more compact.

commit 42cfc8cb (2018-11-27): Merge pull request #275 from eegsynth/issue271

FIX #271 - send triggers and control values more consistently
commit a1ef5c56 (2018-11-27): FIX #271 - send triggers and control values more consistently
closed 2018-11-27 by robertoostenveld