eegsynth/eegsynth #274: the figure of plotspectral has different axes

robertoostenveld, 2018-11-26:
[image]

Both top and bottom one scale along with the location of the red and blue band, but the top one has additional text next to the blue lines which causes the freq axis to differ from the bottom one. I think it would be cleaner to have exactly the same axes in both.

stephenwhitmarsh, 2018-11-26:
you can just change the lines of the top (and bottom) figure to fall within
the normal range :-)

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

robertoostenveld, 2018-11-26:
the scaling should be linked to each other, not both free floating and slightly different

commit 5c371bb5 (2018-12-02): FIX #274, set the x-axis consistently; made plotting of red and blue lines and text optional
closed 2018-12-02 by robertoostenveld