eegsynth/eegsynth #283: generateclock parameters unclear(ly documented)

stephenwhitmarsh, 2018-12-09:
...partly because the function of the module is unclear as ini suggest it does more than generate a clock signal;

- Can we remove the work 'synchronize' in the general input of the ini and replace with 'output' or similar?
- Unclear why, in the offset section of the ini, we have a fraction: multiply=-2.015748031495
- Unclear whether parameter ultimately is in BPM, Hz(Freq) or T(s).
- Why is there a (lower) limit on the output?

stephenwhitmarsh, 2018-12-09:
- adjust parameter?

robertoostenveld, 2018-12-10:
Re 'synchronize': you seem to be looking at an old version of the code, please update.

Re fraction for multiply offset: that is (64/127)*4, i.e. the central position of a launchcontrol knob corresponds exactly to 0.  

Re parameter: fair point. 

Re lower limit: if there is no lower limit, the inter-beat interval can go to nearly infinite with a turn of the knob: which means that it will never ever send a clock signal again... This works badly with the updating: in case of an update, there is currently _not_ a reset of the previously scheduled clock events.  

Re adjust: see readme.

robertoostenveld, 2018-12-10:
I think that the whole module (as many others) need better documentation, and that the ini file is not the place to put or expect that documentation.

stephenwhitmarsh, 2018-12-10:
Re 'synchronize', no it's in the comment of the ini (in latest version).
Re adjust, ah yes. 
I think I know enough now to improve the documentation (readme and ini), as well as for the divide/multiplication module, also starting on an 'events' tutorial.

robertoostenveld, 2018-12-10:
ah, there it is. It should read "boolean, whether to send clock signals over Redis/MIDI".

Note that Redis would always work, but MIDI not (in case no MIDI device is attached). But I think it would actually be nicer to have this as "play/pause/rewind" option, which makes it more consistent with other modules that have play/pause/rewind.

stephenwhitmarsh, 2018-12-10:
Re MIDI, would it not be more consistent to have MIDI devices connected via Redis rather than directly via MIDI? Is there a reason to implement a MIDI interaction within a module?

robertoostenveld, 2018-12-10:
The MIDI clock signal is 24 ticks per quarter note (and needs to be 48 for some), or some 30-40 (up to 100) events per second that have to be precisely timed. That ain't something Redis and another Python module can handle. Normal MIDI notes and controls are sent (e.g. by the sequencer) through Redis.

The idea of this module is that an analog synthesizer (through triggers) and a MIDI synthesizer should be able to synchronize by means of a shared clock. But you can also use it for either one. 

And yes: there are reasons to implement MIDI interactions in multiple modules (as we have discussed multiple times already). Just like there are reasons to implement 5 different normalizations of control signals in all sorts of modules...

stephenwhitmarsh, 2018-12-10:
gotit

robertoostenveld, 2018-12-10:
Have you tried driving your MicroBrute using the clockgenerator and the sequencer?

stephenwhitmarsh, 2018-12-11:
that's the plan, well actually to connect it to the Endorphines Cargo, in
my (user)case :).

robertoostenveld, 2018-12-11:
you mean the [shuttle control](https://www.endorphin.es/assets/endorphin.es_shuttle_control_manual_170327.pdf) right? It is confusing, but I think "Cargo" goes with Endorphines as the name of the manufacturer, not the name of the module.

With the MicroBrute I would expect the EEGsynth/generateclock to control the rate of the built-in step sequencer.
 
What do you expect to happen by connecting EEGsynth/generateclock to the Shuttle Control? It is not clear to me what it does with the MIDI clock signals (but see page 12 and further in the pdf).

I would expect all features of that (i.e. all 16 connectors with CVs and gates) to be controlled by the EEGsynth/endorphines module. So I would implement it as generateclock -> Redis -> endorphines -> MIDI -> shuttlecontrol.

stephenwhitmarsh, 2018-12-11:
Shuttle control, yeah I mix them up all the time.
Pretty sure it can do MIDI synchronization as well and output e.g. random
voltages/LFO at a rate determined by ppm etc. I'll look into it, else I do
the microbrute as you say

stephenwhitmarsh, 2018-12-11:
the new manual is very nice, also explains the MIDI clock on page 12, and
many ways of 'use' that clock with the shuttle control:
https://www.endorphin.es/assets/endorphin.es_shuttle_control_manual_170327.pdf

robertoostenveld, 2018-12-12:
I realized that there is also some variation in MIDI implementations. Most use 24 PPQN, but some use 48. Hence, a possible more clean configuration could be

```
[midi]
device=xxx
ppqn=24 ; can be 24 or 48

[redis]
hostname=xxx
port=xxx
ppqn=1 ; can be 1, 2, 3, 4, 6, 8, 12, 24
shift=xxx
```

robertoostenveld, 2018-12-12:
the previous would of course also imply some changes in the code, so that the master clock runs at the right rate.