eegsynth/eegsynth #286: generate clock hangs when midi device not present

stephenwhitmarsh, 2018-12-11:
Running without proper midi device:

stephenwhitmarsh, 2018-12-11:
Running without proper midi device (to figure out the midi device) makes it hang and not responsive to keyboard interrupt:

stephen@stephen-XPS-13-9360:~/PycharmProjects/eegsynth/patches/rhythms$ python ../../module/generateclock/generateclock.py -i generateclock.ini
------ OUTPUT ------
Midi Through:Midi Through Port-0 14:0
-------------------------
Error: cannot connect to MIDI output
Error: cannot connect to MIDI output
^C^C^C^C^C
^C
^C
^C
^C
^C
^C^C^C^C
^C^C^C^C^C^C

robertoostenveld, 2018-12-12:
You are saying "without proper midi device". But what is in your ini file?

stephenwhitmarsh, 2018-12-12:
This happens whenever the midi device in the .ini points to a non-existing device, e.g. with the wrong port or if the MIDI device is not connected. Other modules would just break and return, in this module it hangs.

robertoostenveld, 2018-12-12:
I can reproduce it. The problem is that the threads (which by that time have started) don't stop properly: the code is calling "exit()" without cleaning up the threads.  I think it can be fixed.

commit 7cca54c7 (2018-12-12): FIX #286 - raise the correct error, don't call exit() bluntly but leave that to the main process (which should stop the threads)
closed 2018-12-12 by robertoostenveld