eegsynth/eegsynth #291: Wirdcard/searchstr for midi devices

stephenwhitmarsh, 2018-12-15:
zoals aan de tel besproken

robertoostenveld, 2018-12-16:
I see on https://mido.readthedocs.io/en/latest/backends/rtmidi.html 

> The ALSA client and port numbers (“128:0” in this case) can change from session to session, making it hard to hard code port names or use them in config files.
>
> To get around this the RtMidi backend allows you to leave out the the port number of port number and client names. These lines will all open the port above:

This might already solve the problem on Linux...

stephenwhitmarsh, 2018-12-16:
hmm i tried that yesterday, but that didn't work

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

robertoostenveld, 2019-08-25:
You can now use a fuzzy match to select one of the MIDI devices.

commit 74ce5cfa (2019-08-25): FIX #291 - support fuzzy match for MIDI devices
closed 2019-08-25 by robertoostenveld