eegsynth/eegsynth #292: improve outputmidi

robertoostenveld, 2018-12-16:
There are two modes of operandi: on the one hand it repeatedly sends continuous values which are get/set in Redis. On the other hand it waits for triggers and sends an updated value whenever it changes. Both have their use cases. Now they are implemented as MIDI "control_change" and "note_on", which are the most obvious match. 

I propose to make the distinction in the ini file on basis of the mechanism (continuous, trigger) and for both mechanisms support a variety of MIDI messages. E.g.

```
[continuous]
control077=launchcontrol.control077 ; i.e. reverse of LaunchControl slider
control078=eeg.alpha
note044=eeg.beta ; if you were to listen to this, it would sound ping-ping-ping-.... at the delay rate

[trigger]
control081= launchcontrol.control081 ; only when slider was moved
note045=eyeblink.channel01  ; only ping on the eye blink itself
start=launchcontrol.control082 ; button
```

This allows to include other MIDI messages, like the ones listed on <https://mido.readthedocs.io/en/latest/message_types.html>

robertoostenveld, 2018-12-16:
here is an updated (work in progress) version of the README. Feel free to comment.

----

# Output MIDI Module

The purpose of this module is to send triggers and control values from Redis to an external MIDI device. The mapping between triggers and control channels can be fully configured in the ini file.

In the EEGsynth we are using two different mechanisms. On the one hand we have continuous control channels using Redis *set/get*, which allow relatively slowly changing values to be represented persistently. On the other hand, we use Redis *publish/subscribe* to trigger precisely synchronized updates.

## Triggers

Under the `[trigger]` section you specify Redis messages that should be sent as MIDI messages at the moment that they change.

You can specify MIDI *note_on* messages using `noteXXX` where XXX is the MIDI note number. For example you can specify `note060` to play [A4 at 440Hz](https://newt.phys.unsw.edu.au/jw/notes.html)).

You can specify MIDI *control_change* messages using `controlXXX` where XXX is the MIDI control channel.

## Continuous control channels

Under the `[control]` section you specify Redis channels that continuously should be sent as MIDI messages, regardless of whether they change. The rate at which the values are read from Redis and sent to MIDI is determined by the general `delay` parameter.

The specification of the mapping between Redis and MIDI is the same as for triggers. If you would specify 'note060' and connect it to a MIDI piano, you would repeatedly hear the C4 being played.

Besides control channels, it is useful to have *start* and *stop* messages sent repeatedly. You could for example set up a patch to start a sequencer if the EEG alpha is high enough, and to stop it again if the EEG alpha is too low.

robertoostenveld, 2018-12-16:
I propose to implement control (=control_change), note (=note_on), polytouch, aftertouch, pitchwheel, start, continue, stop, reset from the list on https://mido.readthedocs.io/en/latest/message_types.htm.

stephenwhitmarsh, 2018-12-16:
Small suggestions for clarity:Triggers

Under the [trigger] section you specify Redis messages that should be sent
as MIDI messages at the moment that they change. -->

Under the [trigger] section you specify both MIDI messages and Redis keys.
The MIDI messages as send as soon as the value of the Redis key changes.

Continuous control channels

The specification of the mapping between Redis and MIDI is the same as for
triggers. If you would specify 'note060' and connect it to a MIDI piano,
you would repeatedly hear the C4 being played. -->

The specification of the mapping between Redis and MIDI is the same as for
triggers. If you would specify 'note060' and connect it to a MIDI piano,
you would repeatedly hear the C4 being played, at the rate of the delay
 parameter.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

commit fe042456 (2018-12-16): FIX #292 - added multiple MIDI messages types to outputmidi module
updated the documentation, and fixed various small issues compared to the previous commit.
closed 2018-12-16 by robertoostenveld