eegsynth/eegsynth #293: improve clockmultiplier

robertoostenveld, 2018-12-16:
Imagine the input at rate X, and the output at rate N*X.

Right now it sends the first output at the input trigger, and then tries to interpolate the next (N-1) ones in between the current and the next (expected) input trigger. If the next input trigger is earlier or later, this results in a very audible mismatch.

Better would be to schedule the output triggers such that last output trigger is at the (expected) next input trigger. This would allow to measure the difference between scheduled and actual incoming trigger and prospectively correct for drift in the input rate by linear interpolation.

stephenwhitmarsh, 2018-12-16:
One can also think of a 'soft sync', in which the rate is gradually up/down
regulated. This is common in hardware modules, perhaps actually for that
reason. A single 'learning-rate' as we have in other places could work.

robertoostenveld, 2020-04-13:
Right now this is not urgent enough to work on

closed 2020-04-13 by robertoostenveld