eegsynth/eegsynth #305: Standardize terminal output

stephenwhitmarsh, 2019-01-03:
Some ideas:
* Boilerplate with license
* Tic-toc-like output to display whether module runs according to intended speed of iterations (as specified in ```delay=``` in ini)
* print input/output redis variables and server information (redis and FT) when ```debug > 0```
* print input/output redis values (at change) when ```debug > 1```
* print message "CNTR-C to exit

robertoostenveld, 2019-01-26:
The `show_change` function that is used in some modules is a bit in this direction. It is a local function, not part of the EEGsynth.py "library"

stephenwhitmarsh, 2019-06-14:
@robertoostenveld, do you have anything to add to the list of requirement here?
Also, "print input/output redis variables " is too undefined: all that might change? all? (which means whole ini file). Perhaps boil it down to a useful - but somewhat arbitrary - selection?

robertoostenveld, 2019-06-14:
the selection of what to print should be module specific. I think we should look into https://docs.python.org/3/library/logging.html

that is similar in concept to FT_ERROR, FT_WARNING, FT_NOTICE, FT_INFO, FT_DEBUG.

robertoostenveld, 2019-07-28:
- [ ] Boilerplate with license
- [ ] Tic-toc-like output to display whether module runs according to intended speed of iterations (as specified in delay= in ini)
- [ ] print input/output redis variables and server information (redis and FT) when debug > 0
- [ ] print input/output redis values (at change) when debug > 1
- [ ] print message "CNTR-C to exit

closed 2019-07-28 by robertoostenveld

robertoostenveld, 2019-07-28:
using `monitor.loop()` just after the `while True`, it now consistently prints 

```
Starting loop...
looping with 18 iterations in 1.00681 seconds
looping with 18 iterations in 1.00145 seconds
looping with 19 iterations in 1.05474 seconds
```

robertoostenveld, 2019-07-28:
printing of (redis) values when they have changed is implemented just like with the old `show_change()` function. The new `monitor.update()` function has a debug option, kust like `patch.setvalue()`

robertoostenveld, 2019-07-28:
the only thing that I did not implement is the printing of server information (Redis and FT). Also, the new functionality should still be more extensively adopted in the existing code.

stephenwhitmarsh, 2019-07-29:
Super, great idea for the show_change(). It makes it easy to implement in
every module.  But, to be clear, it will display all changes to redis, not
only those done by particular function, right? It makes one wish Redis
would have a dedictated field for the 'operator' so we could filter for who
made the Redis requests.

robertoostenveld, 2019-07-29:
It does not monitor Redis per see. It monitors values of local variables in Python (which may come from Redis). 

Perhaps best explaind with an example...:

```
patch = EEGsynth.patch(config, redis) # as before
monitor = EEGsynth.monitor()   # prints boiler plate

while True:
  monitor.loop()    # prints number of iterations per second (approximately)
  time.sleep(0.05)  # wait for the given delay period

  # the following is pretty standard
  key = 'something'
  scale = patch.getfloat('scale', key)
  offset = patch.getfloat('offset', key)
  val = patch.getfloat('input', key)
  val = EEGsynth.rescale(val, slope=scale, offset=offset)

  # prints it, but only when different from previous value
  monitor.update(key, val)

  # do something with it
  # ...
```

Furthermore, it is also possible (partially that was already possible, partially it is newly implemented) to do 

```
# this comes from the config.ini or from Redis
val = patch.getfloat('section', key, debug=True)
val = patch.getint('section', key, debug=True)
val = patch.getstring('section', key, debug=True)

# this goes to Redis
patch.setvalue(key, val, debug=True)  
```

where in the latter case it always gets printed, i.e. either when reading or writing the value. The `monitor` object is smarter and keeps track of the previous value (in a dict). But the `getxxx` and `setvalue` methods of the patch allow you to see all changes that relate to the specific module. In general I find that too noisy. 

Oh, and note that the monitor object itself does not know about the specific debug state, but you can use it like this in the module code:

```
monitor.update(key, val, debug=(debug>0))  # for level 1 debugging
monitor.update(key, val, debug=(debug>1))  # for level 2 debugging
monitor.update(key, val, debug=(debug>2))  # for level 3 debugging
```

This is actually the same as for the patch `getxxx` and `setvalue` methods, except that those have by default a `debug` value that is `False`, whereas `monitor.update()` by default assumes `debug` to be `True`.

stephenwhitmarsh, 2019-07-29:
Ah, I see! I thought it was a single call in the beginning of the code, but
we have to implement it whenever we want to update the values in the
terminal (with monitor taking care to only do it with a change). That makes
sense and visible and flexible, but yes we have to add it consistently.

commit a223cfb7 (2019-07-28): FIX #305 - implemented EEGsynth monitor class, this replaces show_change and prints a boilerplate with license